victronenergy/node-red-contrib-victron #7: Show connection status messages on nodes

pkronstrom, 2019-01-08:
Node-RED supports showing connection status below the nodes. Show a red 'disconnected' or green 'connected' circle below the node based on the dbus connection status.

mpvader, 2019-01-10:
A connection loss to D-Bus entirely is not a use case to cater for; as discussed in #11.

But; a loss of connection to one of the devices could be a viable option. Consider this use case:

1. User configures a system; with a battery on VE.Direct Port 1.
2. Stores it
3. Modifies if (plugs  the BMV to VE.Direct port 2; which changes the device instance).
4. Opens the node-red ui again

Now, in the flow, it would be great that the items he has in his flow show a connection status below the nodes.

pkronstrom, 2019-01-23:
Implementation in commit https://github.com/victronenergy/node-red-contrib-victron/commit/14a9ff4ff907ee99d312672a9af8b57ecbe536f2

The node connection status is shown for the currently selected service (and path). When a service disconnects or a service comes back online, the node status is updated accordingly.

[image]

closed 2019-01-23 by pkronstrom