victronenergy/node-red-contrib-victron #9: Add an option for native dbus instead of using dbus over TCP

pkronstrom, 2019-01-08:
Using the dbus over TCP works well on external node-red installations. This extension should connect directly to the system dbus on Venus installations in order to skip the dbus-over-tcp configuration step.

mpvader, 2019-01-08:
Note that running node-red external is nice for your development now - only. It will not be a use case later on.

mpvader, 2019-01-10:
dbuslistener already has (working) code for this, its here:
https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/services/dbus-listener.js#L31

Same functionality is used in the [Venus-beta-builds with Signalk](https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/services/dbus-listener.js#L31).

pkronstrom, 2019-01-16:
Changes in https://github.com/victronenergy/node-red-contrib-victron/commit/8b438e9c344cbb8a1bcf02868ddeae29c88171c6

closed 2019-01-16 by pkronstrom