victronenergy/node-red-contrib-victron #11: Reconnect the client if the connection was lost

pkronstrom, 2019-01-09:
Currently, the dbus-listener.js does not reconnect, if the connection to the system dbus was lost for some reason. This will cause the victron energy nodes not to function in case the dbus service is restarted for some reason (or dbus service started after node-red, for example).
 
See
https://github.com/sbender9/signalk-venus-plugin/issues/7
and
https://github.com/sbender9/signalk-venus-plugin/blob/master/dbus-listener.js

mpvader, 2019-01-09:
Note that; as in real life your code will run on the venus device; this will never happen.

There is no other oroject on Vneus OS that survives a dbus reset.

Instead; we have a watchdog that restarts the full venus device/os upon a dbus-daemon crash/reset

closed 2019-01-09 by pkronstrom

wollew, 2023-06-05:
> Note that; as in real life your code will run on the venus device; this will never happen.

I know this is an old issue, but I just ran into this issue. Your statement is only true if you're using this on the Venus device itself, if you deploy them to an external node-red instance then this can and will happen, a simple reboot of the Venus device is enough to trigger the problem.