victronenergy/node-red-contrib-victron #12: Show a disclaimer in Relay node, if the relay is not on manual operation

pkronstrom, 2019-01-09:
If the relay is set on, e.g. Generator Mode, show a disclaimer to the end user on node edit dialog.

The MVP solution for this would be to just show a warning text in relay node edit dialog:
"Please ensure that the relay is set to 'Manual' in settings (Relay -> Function)"

pkronstrom, 2019-01-09:
_Can the relay still be toggled via dbus, even if it's set on generator use?_
In this case, one option would be to show both the warning message, and a greyed out relay option, for example.
-> Where can I see the current relay function?

mpvader, 2019-01-10:
See gui code for how that function works. Its a setting.

Try toggling the relay when set to generator mode

pkronstrom, 2019-01-21:
Implementation in https://github.com/victronenergy/node-red-contrib-victron/commit/8c4071038f96b82bfe8d6c02f9a8a0718649584e

The user is shown a warning message if the relay is reserved for some other function than manual.

[image]

Trying to trigger the relay sends also a warning to the debug stream. The relay will not trigger in this case.

[image]

closed 2019-01-21 by pkronstrom