victronenergy/node-red-contrib-victron #33: Use VRM API to get d-bus path data instead of manual table exports

mpvader, 2019-02-13:
Once #32 is completed; we can look into this; so its no longer necessary to now and then make an export from the VRM Database.

mpvader, 2019-02-13:
```https://vrmapi.victronenergy.com/v2/admin/data-attributes```

The rest (dataAttributeEnums) is not available yet; and will be made available later.

dirkjanfaber, 2021-09-10:
We might need to separate use-cases between dbus and modbus_tcp usage. Via modbus_tcp not all of the services are available.

By default (dbus connection) show all of the VRM attributes. When using modbus_tcp, only show the services that have the  'modbus_tcp' flag set.

We could use https://github.com/victronenergy/dbus_modbustcp/blob/master/CCGX-Modbus-TCP-register-list.xlsx as a source. But that beats a bit of the idea of only using the VRM API.

dirkjanfaber, 2024-11-08:
Closing this as we are using the dbus_modbustcp list now, which works good enough.

closed 2024-11-08 by dirkjanfaber