victronenergy/node-red-contrib-victron #37: Improve no devices found text

mpvader, 2019-03-10:
Below screenshot shows what you see after dragging in a node for which there is no matching device on the system. Lots of people will see it; for many it will even be the first thing they see; at least it was like that for me :).

Change the text to `No matching devices found. Pick a different node from the palette.`, or something better.

[image]

pkronstrom, 2019-03-12:
It might be better to show disabled dropdowns and a message below them instead of a big scary yellow box. Also, specifying the device type in the message is less ambiguous.

Here's a suggestion:

On a new node, the selects are disabled and a red warning text.
[image]

When a previously selected device appears disconnected, the selected values are shown, but not editable. A different warning text is shown.

[image]

Changes [here](https://github.com/victronenergy/node-red-contrib-victron/commit/a62dcfb1785afbb7f933aa4cd8c651260501013a)

pkronstrom, 2019-03-12:
New style

[image]

pkronstrom, 2019-03-12:
New style

[image]

pkronstrom, 2019-03-13:
Merged to master

closed 2019-03-13 by pkronstrom