victronenergy/node-red-contrib-victron #38: Improve the naming

mpvader, 2019-03-10:
Right now we only use the product name in the pulldown (see screenshot). Adding the connection or something could help. Perhaps use a dual line per entry html pulldown, so the connection can be on its own line?

```
root@beaglebone:~# dbus -y  com.victronenergy.temperature.builtin_adc3_di1 /Mgmt/Connection GetValue
value = 'Temperature sensor input 2'
```

[image]

[image]

pkronstrom, 2019-03-12:
It is indeed a challenge to identify services from each other (for a user, that is). Here's a few thoughts on this.

Currently a custom name is shown for each device -- if it is not set, the dropdown label falls back to the product name, which leads to ambiguity in the dropdowns.

I don't really think that showing the dbus service identifier would work here, since the user can't use ttyO0 / ttyO1 / ttyO2 etc. to actually identify similar  devices from each other. If the user actually has several devices of the same type, I would just encourage them to add a custom name from the settings.

Using the /Mgmt/Connection might work as a solution, but is it common for similar devices to use different connection methods? I fear, that for e.g. battery monitors the labels would still stay indistinguishable: BMV-700 (VE.Direct), BMV-700 (VE.Direct), etc. Yet again, setting a CustomName works here.

pkronstrom, 2019-03-12:
Something like this is should be quite quick to add, if we want to show some service-specific data for the selected service.

[image]

mpvader, 2019-03-12:
Ok;

how are people going to know that setting up a custom name is a solution to their problem?

mpvader, 2019-03-12:
Hm; cross post. I’ll remove my previous comment.

Ok that will help. But; please do check whats under /Mgmt/Connection for a bmv on a VE.Direct port. Is it really just “VE.Direct”? No port number, nothing?

I’d preft not to show the service name

pkronstrom, 2019-03-12:
> Ok that will help. But; please do check whats under /Mgmt/Connection for a bmv on a VE.Direct port. Is it

For both simulations and our physical BMV-712: "/Mgmt/Connection": "VE.Direct"

mpvader, 2019-03-12:
Ok thats too bad.

Then, what about just showing a tip, and only show it if multiple options in the dropdown have the same exact string as a name?

“Tip: go into the menus and set a Custom name to easily distinguish one device from the other.”

pkronstrom, 2019-03-12:
The following tip is shown if multiple devices with the same name are shown in the edit UI.

[image]

mpvader, 2019-03-12:
👍

pkronstrom, 2019-03-13:
Merged to master

closed 2019-03-13 by pkronstrom