victronenergy/node-red-contrib-victron #40: See if we can proxy node-red via hiawatha

mpvader, 2019-03-15:
The goal is to host it under http://venus.local/node-red, rather than http://venus.local:1880/.

Might be hard to do because you then also need to proxy websockets. If hard; then never mind; and at a minimum we can consider putting in a redirect in hiawatha.

If easy; then nice.

tkurki, 2019-03-18:
Hiawatha has ws support.

pkronstrom, 2019-03-19:
Alrighty, the following changes are needed to get it working.

1. Create a config file for Hiawatha in `/etc/hiawatha/sites-enabled/node-red.conf`.
Add the following line `ReverseProxy ^/node-red http://127.0.0.1:1880/ Keep-Alive`. Restart Hiawatha `/etc/init.d/hiawatha restart`

2. Add the following line to `/home/root/.node-red/settings.js`: `httpAdminRoot: '/nodered'`.

3. Change [this row](https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/nodes/victron-nodes.html#L47) to accommodate the httpAdminRoot property in settings.js.
```
const serviceRoot = (RED.settings.httpNodeRoot || RED.settings.httpAdminRoot || "").replace(/\/$/,"") + '/victron/services/';
```

This works OK, but occasionally I've been getting some lag spikes in node-red websocket data. I don't really think that it is actually Hiawatha being the root cause, though. I'll keep testing it a bit.

pkronstrom, 2019-03-21:
> 3. Change [this row](https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/nodes/victron-nodes.html#L47) to accommodate the httpAdminRoot property in settings.js. I'll commit this change to the master soon.

This is now included in https://github.com/victronenergy/node-red-contrib-victron/commit/e7326b3c64b7dcf0e884557232a08a3e1c24c5f0

pkronstrom, 2019-03-21:
I'm not convinced that the reverse-proxy will work without hiccups. Occasionally the whole Node-RED UI freezes and reloading it takes ages -- usually when a lot of requests are triggered simualtenously (e.g. repeatedly triggering an inject node).

This was discussed with Matthijs, let's do a redirect instead.

pkronstrom, 2019-03-27:
Hiawatha can't be used to redirect to port when the exact IP isn't known (see Hugo's answer [here](https://www.hiawatha-webserver.org/forum/topic/217)). This answer is 10 years old, though.

A simple workaround serves a php script that redirects from /nodered to :1880/nodered. `httpRoot`needs to be set in settings.js for this to work.

*~/.node-red/settings.js*

```
httpRoot: '/nodered',
```

*/var/www/venus/nodered/redirect.php*

```
<?php
    // redirects /nodered to <ip>:1880/nodered
    $port = '1880';
    header('Location: '
        . ($_SERVER['HTTPS'] ? 'https' : 'http')
        . '://' . $_SERVER['HTTP_HOST'] . ':' . $port
        . $_SERVER['REQUEST_URI']);
    exit;
?>
```

*/etc/hiawatha/sites-enabled/nodered.conf*

```
UrlToolkit {
    ToolkitID = rewrite
    Match ^/nodered/(.*) Rewrite /nodered/redirect.php
}

UseToolkit = rewrite
```

dirkjanfaber, 2021-09-09:
Having 2 urls (with and without adding 1880)  to open node-red instead of just one probably confuses more than it solves.

The Proxy solution would have been nice, but that still gives some strange hickups (just tested). So we won't add the functionality to Venus for now. Closing this issue.

closed 2021-09-09 by dirkjanfaber