victronenergy/node-red-contrib-victron #45: Connection interrupted briefly but node remains inoperative

svwhisper, 2020-01-15:
I'm running node red 1.03 on Ubuntu connecting to CCGX via TCP.  On node red start, all is good and messages are emitted from a victron-input-vebus node configured to receive temp and overload alarms from my Quattro.  Occasionally, incoming messages cease and the error log contains the messages below.  I suspect this is because of a brief interruption of the network between the node red box and the CCGX.  The node continues to show "green" in the GUI, though clearly it isn't truly connected.

When a read timeout occurs, the connection should ideally be taken down and connection retried repeatedly until successful.

These nodes are very useful, so thanks for creating them.  My initial use case is to intelligently shed loads in a remote off-grid home based on system overload, high temp or state of charge.

Error connecting to dbus: Error: read ETIMEDOUT
Error connecting to dbus: Error [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed
Error connecting to dbus: Error [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed
Error connecting to dbus: Error [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed

farmcoding, 2020-05-15:
I'm running v1.0.5 on Rasbian, connecting to Venus device via TCP - same issue from the description above. Let me know if I can run anything to give more details?

mpvader, 2020-05-15:
hi both - we'll look into this. But not too urgently I'm afraid, our main focus is to add node-red and these nodes into our Venus OS images. And in that situation we don't above issue. More information about that [here](https://www.victronenergy.com/live/venus-os:large-image).

In case you find & fix the issue yourself, pull requests welcome ofcourse.

davenrea, 2020-12-10:
has anyone got any further with this issue as I am seeing it often on rasbian 1.2.6 .

thanks

Dave

timdoyle91, 2021-03-02:
I've been having this issue since I set up my Raspberry Pi server connected to my CCGX. I'm assuming the only workaround is to periodically restart node-red or restart based on the error? Restarting the flow itself doesn't seem to sort it.

I don't want to fully deploy my setup until I've sorted this as there's a chance it could turn on my solar dump, lose the connection and then not turn it back off, which would damage my batteries.

dirkjanfaber, 2021-09-09:
Does the problem still exist? There have been several updates on both Node-RED and the devices firmware since the last report.

If so, is the common denominator a CCGX? And which firmware version does it have? In order to reproduce (and fix this), this information is essential.

dirkjanfaber, 2021-09-13:
Closing this issue. If the problem still exists on the latest version of the firmware, please create a new issue.

closed 2021-09-13 by dirkjanfaber