victronenergy/node-red-contrib-victron #46: External Node-red connecting to multiple GX devices

RobertSmart, 2020-01-31:
Hi,

I'm just having a play with this and would like to get information from more than one GX device on the same network. 

After a brief look there doesn't seem to be a way to do this as the DBUS address is hardcoded in the env_variables.

Is there a way to do this currently or are there any other work arounds?

Thanks,

R Smart

sbender9, 2020-04-20:
@RobertSmart I'll look into this. Probably won't be simple, but should be doable.

mpvader, 2020-04-21:
To add, whats the use case @RobertSmart ? What sort of victron products will be connected to each gx device?

If its for example tank levels, then an easier solution will be to hook the two GX-es up using VE.Can.

mpvader, 2021-01-06:
closing, no explanation for why we should be doing this (welcome to re-open if there is a real good use case)

closed 2021-01-06 by mpvader