victronenergy/node-red-contrib-victron #56: New 1.3.0 version does not load pallets, non functioning

mbosys, 2020-04-28:
tried installing new version but does not even load icons into node red, and has missing files on install.
Now broken 

I am also trying to get previous versions working on hassio node red container install but no service issue on any loaded icons have dbus connected in log, but will not function, if anyone can help?

sbender9, 2020-04-28:
I think you may have the old version still installed. Do you have ~/.node-red/node_modules/node-red-contrib-victron ? Delete that.

mbosys, 2020-04-28:
Thanks for getting back! no is the new directory. node_modules/@victronenergy/node-red-contrib-victron.
Don't think its ever been installed without the @victronenergy before on my system.

tried on the node red pallet install, and fully removed, also via npm install via ssh, here is the log of errors from that install

"53 verbose stack Error: ENOENT: no such file or directory, rename '/usr/share/hassio/homeassistant/node-red/node_modules/.staging/@victronenergy/node-red-contrib-victron-889e0799/node_modules/debug' -> '/usr/share/hassio/homeassistant/node-red/node_modules/.staging/debug-e102e51d'
154 verbose cwd /usr/share/hassio/homeassistant/node-red/node_modules
155 verbose Linux 4.15.0-96-generic
156 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "@victronenergy/node-red-contrib-victron"
157 verbose node v14.0.0
158 verbose npm  v6.14.4
159 error code ENOENT
160 error syscall rename
161 error path /usr/share/hassio/homeassistant/node-red/node_modules/.staging/@victronenergy/node-red-contrib-victron-889e0799/node_modules/debug
162 error dest /usr/share/hassio/homeassistant/node-red/node_modules/.staging/debug-e102e51d
163 error errno -2
164 error enoent ENOENT: no such file or directory, rename '/usr/share/hassio/homeassistant/node-red/node_modules/.staging/@victronenergy/node-red-contrib-victron-889e0799/node_modules/debug' -> '/usr/share/hassio/homeassistant/node-red/node_modules/.staging/debug-e102e51d'
165 error enoent This is related to npm not being able to find a file.
166 verbose exit [ -2, true ]"

[image]

Pallet is there but no live modules.

Managed to installer older version and modules become live again.  but still no service on entering any modules on node red

Tried on another install of node red on different system and same for 1.3.0 no live modules on pallet.

mbosys, 2020-04-29:
Will leave the issue with the new 1.3.0 loading pallets on here, and split my issue on connections to here.

https://community.victronenergy.com/questions/35001/external-node-red-on-ubuntu-connecting-to-ccgx.html?childToView=48325#answer-48325

sbender9, 2020-04-29:
Strange. Now I’m able to reproduce your issue. I’m getting the same now on a fresh install. Should have a fix today...

mbosys, 2020-04-29:
Thanks Scott, glad it was not just my issue! and sorry for the extra work!

sbender9, 2020-04-29:
Can you please try with v1.3.1, should fix the issue. Although I'm still not sure about those install errors...

#57

mbosys, 2020-04-29:
Thanks Scott, install's ok now in node red the pallet is back!

closed 2020-04-30 by sbender9