victronenergy/node-red-contrib-victron #62: Cannot change state of Relay for MPPT Charge Controller

KidA001, 2020-05-21:
I'm running Node-Red on my rPi VenusOS which is connected to my Victron 150/100 Charge Controller and Battery Shunt. I created a flow to open or close the relay on the charge controller but noticed the state of the relay wasn't actually changing.

I'm debugging the flow by logging the msg.payload I'm sending to the charge controller relay, and also by logging the current state of the relay. Regardless of what I send to the relay it does not change its state. The message I'm sending to the relay is either `{ payload: 1 }` or `{ payload: 0 }`

Either I'm sending the wrong payload, the victron app is overwriting what I'm sending, or the node-red-contrib-victron can't properly change relay state. Below is a screenshot of my flow along with the JSON. I've left the debug's in there

[image]

```javascript
[{"id":"e4adfc06.24e05","type":"tab","label":"Open Relay based on Temp","disabled":false,"info":""},{"id":"5f535281.ce772c","type":"openweathermap in","z":"e4adfc06.24e05","name":"Current Temperature","wtype":"current","lon":"","lat":"","city":"Nevada City","country":"United States","language":"en","x":190,"y":300,"wires":[["af7e6b3a.3e6e88","ddad3658.6ea458"]]},{"id":"af7e6b3a.3e6e88","type":"function","z":"e4adfc06.24e05","name":"If temp above 18c open relay","func":"if (msg.payload.tempc > 18) {\n    return { payload: 1 };\n} else {\n    return { payload: 0 };\n}","outputs":1,"noerr":0,"x":460,"y":300,"wires":[["ddad3658.6ea458","5ab5375e.9063b8"]]},{"id":"5ab5375e.9063b8","type":"victron-output-relay","z":"e4adfc06.24e05","service":"com.victronenergy.solarcharger.ttyUSB1","path":"/Relay/0/State","serviceObj":{"service":"com.victronenergy.solarcharger.ttyUSB1","name":"SmartSolar Charger MPPT 150/100 rev2","paths":[{"path":"/Relay/0/State","type":"enum","name":"Relay on the charger","enum":{"0":"Open","1":"Closed"},"writable":true}]},"pathObj":{"path":"/Relay/0/State","type":"enum","name":"Relay on the charger","enum":{"0":"Open","1":"Closed"},"writable":true},"initial":"","name":"","x":870,"y":300,"wires":[]},{"id":"ddad3658.6ea458","type":"debug","z":"e4adfc06.24e05","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":450,"y":220,"wires":[]},{"id":"ba08535e.c8a07","type":"debug","z":"e4adfc06.24e05","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":650,"y":400,"wires":[]},{"id":"66c5df24.5966a","type":"victron-input-solarcharger","z":"e4adfc06.24e05","service":"com.victronenergy.solarcharger.ttyUSB1","path":"/Relay/0/State","serviceObj":{"service":"com.victronenergy.solarcharger.ttyUSB1","name":"SmartSolar Charger MPPT 150/100 rev2","paths":[{"path":"/Dc/0/Voltage","type":"float","name":"Battery voltage (V)"},{"path":"/Dc/0/Current","type":"float","name":"Battery current (A)"},{"path":"/Mode","type":"enum","name":"Charger on/off","enum":{"0":"Off (deprecated)","1":"On","4":"Off"}},{"path":"/State","type":"enum","name":"Charge state","enum":{"0":"Off","2":"Fault","3":"Bulk","4":"Absorption","5":"Float","6":"Storage","7":"Equalize","252":"ESS"}},{"path":"/Pv/V","type":"float","name":"PV voltage"},{"path":"/Pv/I","type":"float","name":"PV current"},{"path":"/Relay/0/State","type":"enum","name":"Relay on the charger","enum":{"0":"Open","1":"Closed"}},{"path":"/ErrorCode","type":"enum","name":"Error code","enum":{"0":"No error","1":"#1 - Battery temperature too high","2":"#2 - Battery voltage too high","3":"#3 - Battery temperature sensor miswired (+)","4":"#4 - Battery temperature sensor miswired (-)","5":"#5 - Battery temperature sensor disconnected","6":"#6 - Battery voltage sense miswired (+)","7":"#7 - Battery voltage sense miswired (-)","8":"#8 - Battery voltage sense disconnected","9":"#9 - Battery voltage wire losses too high","17":"#17 - Charger temperature too high","18":"#18 - Charger over-current","19":"#19 - Charger current polarity reversed","20":"#20 - Max Bulk-time exceeded","22":"#22 - Current sensor issue","23":"#23 - Charger temperature sensor disconnected","34":"#24 - PV over current","38":"#38 - Input shutdown due to battery over-voltage","39":"#39 - Input shutdown due to battery over-voltage","67":"#67 - No BMS","114":"#114 - CPU temperature to high","116":"#116 - Calibration data lost","119":"#119 - Settings data lost"}},{"path":"/Load/State","type":"enum","name":"Load state","enum":{"0":"Off","1":"On"}},{"path":"/Yield/User","type":"string","name":"Yield since reset (kWh)"},{"path":"/Yield/Power","type":"float","name":"Total yield (kWh)"},{"path":"/Yield/System","type":"string","name":"Yield since last update (kWh)"},{"path":"/MppOperationMode","type":"enum","name":"MPP operation mode","enum":{"0":"Off","1":"Voltage or current limited","2":"MPPT Tracker active"}}]},"pathObj":{"path":"/Relay/0/State","type":"enum","name":"Relay on the charger","enum":{"0":"Open","1":"Closed"}},"initial":"","name":"","x":320,"y":400,"wires":[["ba08535e.c8a07"]]}]
```

mpvader, 2020-05-21:
Hi, we’ll look at that asap - but might take a few days. You also posted the same question on community right? Can you add a link to that here to prevent double work?

Thank you, Matthijs

KidA001, 2020-05-21:
@mpvader thanks, appreciate the time you all put into developing this.

Yes forum post is [here](https://community.victronenergy.com/questions/50216/relay-on-charge-controller-not-changing-with-node.html)

KidA001, 2020-05-25:
A member in the forum said that "Node red will not control relays on charge controls or any Aux relays on the Multis any time soon" - Just confirming this before I close?

mpvader, 2020-05-25:
Hi, took me a while, but I know now what the issue is. I've documented it all here:

https://www.victronenergy.com/live/venus-os:extended#controlling_relays

So, relay control for MPPTs is coming - I'm not sure when yet. Depends when someone has time to fix VictronConnect to make it possible to set the MPPT relay to be manually controlled

KidA001, 2020-05-27:
@mpvader is the VictronConnect code open source, I haven't seen a repo so I figured not? I'm glad to contribute if possible.

mpvader, 2020-05-27:
Hi. No its not @KidA001

KidA001, 2020-05-27:
@mpvader I'll close the issue. If there is any way I can help or post elsewhere to get support for the victronconnect app to work on this let me know. thanks again for your work on this project

closed 2020-05-27 by KidA001