victronenergy/node-red-contrib-victron #66: NodeRED 1.2.2 Editor stuck at “Adding Flows to workspace

fredsparkle, 2020-10-31:
Did a recent upgrade to one of my Raspberry Pi's to the latest version of NodeRed 1.2.2 and when you open a virgin browser window to port 1880 it goes through it's normal startup until it hangs on the progress message "Adding Flows to workspace".

If I uninstall the node-red-contrib-victron then the editor will fully load with a warning about the missing nodes.

More detail on this issue (including problem flow) is at: 

[https://discourse.nodered.org/t/editor-stuck-at-adding-flows-to-workspace/34306/5](url)

paulguru, 2020-11-01:
I had the same issue when I updated Node Red to version 1.2.2 (i'm using Docker on Raspberry PI). 
Seems that "@victronenergy/node-red-contrib-victron" is not compatible with Node Red 1.2.2.

knolleary, 2020-11-02:
In Node-RED 1.2.x we changed how flows are imported. We introduced detection for importing duplicate nodes - throwing an error when we detect it.

When importing from the clipboard, that error is caught internally and is used to ask the user how to resolve.

However this node is causing the error to be triggered when doing the initial flow load. This is because the `victron-client` config node forces itself into the user's flows whenever the node is added to the palette - https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/nodes/config-client.html#L31

When the editor is loading, it loads the nodes first and then loads the user's flows.  This means, at the point in time `onpaletteadd` is called, the user's flows have not yet been loaded. The node forces a `victron-client` node into the active flow configuration with an id of `victron-client-id`. We then load the user's flows, which already has the `victron-client-id` node in it - causing the duplicate node error to be hit.

In 1.2.3 (to be released tomorrow), we've improved the error reporting so the editor doesn't get stuck at 'adding flows to workspace' - but it does still report this as an error to the user. We can't continue to load the flows with the duplicate ID.

From our perspective, this node ought to be updated to only add the `victron-client` node when its needed - in other words, have an `onadd` function on the HTML definition of all the regular nodes that checks for the required `victron-client-id` node and adds it at that point in time. That will ensure the `victron-client` config node doesn't get added until its actually needed.

mpvader, 2020-11-03:
Hi @knolleary , thats an honor to have the creater of node-red responding here, welcome! Its a really cool fit (Victron Energy / Venus OS / Node-RED) and still in its early paces and proving a challenge to pull it all together; but I'm expecting a lot of this).

Noted the issue, also @fredsparkle & @paulguru , and I've asked Scott to look into it and make the necessary changes.

sbender9, 2020-11-03:
All good here. I will take a look...

-=Scott

davenrea, 2020-11-30:
> Hi @knolleary , thats an honor to have the creater of node-red responding here, welcome! Its a really cool fit (Victron Energy / Venus OS / Node-RED) and still in its early paces and proving a [...]

Do we have an estimated time when this will be resolved? Or a work around?

Thanks

Dave

mpvader, 2021-01-13:
I updated the node-red version to v1.2.7 in the latest image (v2.70~3-large-10). There still seem to be issues, as reported by Bob:

> Yesterday I installed the large image 10 on a Cerbo GX. I just test some simple flows controlling both relays on the Cerbo GX, which worked perfect. When I this morning tried to access the Cerbo Node-Red again it give following error (where I can't seem to get out of):
[image]

closed 2021-01-13 by sbender9

michatroniker, 2021-01-14:
Hi,

we are also facing the issue with "duplicate nodes" - when will the fix for this be released in an .swu image?

thanks,
Michael Schmid (BOS-AG)

mpvader, 2021-01-14:
Hi, for swu images see the node-red issue here: https://github.com/victronenergy/venus/issues/378

SergioRius, 2021-02-23:
My node-red is throwing the duplicate node error and doesn't load the flows. Is offline.
Palette says that the node is used and can't be uninstalled.
How I can revert and get my system back online?
How can a node package that throws the system offline only by installing still be published?

SergioRius, 2021-02-23:
I saw the victron client configuration node in the left pane, so I deleted it and published only modified nodes.
And TADA! I lost all my flows! My flows file is empty and there was still lost of work pending to go into the scheduled backup.
Nice! Well done!

davenrea, 2021-02-23:
> I saw the victron client configuration node in the left pane, so I deleted it and published only modified nodes.
> And TADA! I lost all my flows! My flows file is empty and there was still lots of work pending to go into the scheduled backup.
> Nice! Well done!

I have solved this using a separate Pi and sending all the information from the victron to the mqtt server on the pi, works well. I can send the other end if you need it.

`[
    {
        "id": "993abf19.c055a",
        "type": "tab",
        "label": "Dashboard clone",
        "disabled": false,
        "info": ""
    },
    {
        "id": "24fe5dc3.9ac0f2",
        "type": "victron-input-vebus",
        "z": "993abf19.c055a",
        "service": "com.victronenergy.vebus.ttyO5",
        "path": "/Mode",
        "serviceObj": {
            "service": "com.victronenergy.vebus.ttyO5",
            "name": "Quattro 48/15000/200-2x100",
            "paths": [
                {
                    "path": "/Ac/ActiveIn/L1/V",
                    "type": "float",
                    "name": "Input voltage phase 1 (VAC)"
                },
                {
                    "path": "/Ac/ActiveIn/L2/V",
                    "type": "float",
                    "name": "Input voltage phase 2 (VAC)"
                },
                {
                    "path": "/Ac/ActiveIn/L3/V",
                    "type": "float",
                    "name": "Input voltage phase 3 (VAC)"
                },
                {
                    "path": "/Ac/ActiveIn/L1/I",
                    "type": "float",
                    "name": "Input current phase 1 (A)"
                },
                {
                    "path": "/Ac/ActiveIn/L2/I",
                    "type": "float",
                    "name": "Input current phase 2 (A)"
                },
                {
                    "path": "/Ac/ActiveIn/L3/I",
                    "type": "float",
                    "name": "Input current phase 3 (A)"
                },
                {
                    "path": "/Ac/ActiveIn/L1/F",
                    "type": "float",
                    "name": "Input frequency 1 (Hz)"
                },
                {
                    "path": "/Ac/ActiveIn/L2/F",
                    "type": "float",
                    "name": "Input frequency 2 (Hz)"
                },
                {
                    "path": "/Ac/ActiveIn/L3/F",
                    "type": "float",
                    "name": "Input frequency 3 (Hz)"
                },
                {
[... 5322 lines cut ...]
        "serviceObj": {
            "service": "com.victronenergy.solarcharger.ttyUSB0",
            "name": "SmartSolar Charger MPPT 150/100 rev2",
            "paths": [
                {
                    "path": "/Dc/0/Voltage",
                    "type": "float",
                    "name": "Battery voltage (V)"
                },
                {
                    "path": "/Dc/0/Current",
                    "type": "float",
                    "name": "Battery current (A)"
                },
                {
                    "path": "/Mode",
                    "type": "enum",
                    "name": "Charger on/off",
                    "enum": {
                        "0": "Off (deprecated)",
                        "1": "On",
                        "4": "Off"
                    }
                },
                {

SergioRius, 2021-02-23:
Thanks @davenrea you're so kind. Unfortunately I have some many modifications since the last backup. Being a whole flow with the Eastron modbus processing, reposting to mqtt and related dashboards.
I don't know if it's possible to recover that flows. Like nodered maintaining a flows previous version backup elsewhere.

SergioRius, 2021-02-23:
> > I saw the victron client configuration node in the left pane, so I deleted it and published only modified nodes.
> > And TADA! I lost all my flows! My flows file is empty and there was still lots of work pending to go into the scheduled backup.
> > Nice! Well done!
> 
> I have solved this using a separate Pi and sending all the information from the victron to the mqtt server on the pi, works well. I can send the other end if you need it.
> 
> `[ { "id": "993abf19.c055a", "type": "tab", "label": "Dashboard clone", "disabled": false, "info": "" }, { "id": "24fe5dc3.9ac0f2", "type": "victron-input-vebus", "z": "993abf19.c055a", "service": "com.victronenergy.vebus.ttyO5", "path": "/Mode", "serviceObj": { "service": "com.victronenergy.vebus.ttyO5", "name": "Quattro 48/15000/200-2x100", "paths": [ { "path": "/Ac/ActiveIn/L1/V", "type": "float", "name": "Input voltage phase 1 (VAC)" }, { "path": "/Ac/ActiveIn/L2/V", "type": "float", "name": "Input voltage phase 2 (VAC)" }, { "path": "/Ac/ActiveIn/L3/V", "type": "float", "name": "Input voltage phase 3 (VAC)" }, { "path": "/Ac/ActiveIn/L1/I", "type": "float", "name": "Input current phase 1 (A)" }, { "path": "/Ac/ActiveIn/L2/I", "type": "float", "name": "Input current phase 2 (A)" }, { "path": "/Ac/ActiveIn/L3/I", "type": "float", "name": "Input current phase 3 (A)" }, { "path": "/Ac/ActiveIn/L1/F", "type": "float", "name": "Input frequency 1 (Hz)" }, { "path": "/Ac/ActiveIn/L2/F", "type": "float", "name": "Input frequency 2 (Hz)" }, { "path": "/Ac/ActiveIn/L3/F", "type": "float", "name": "Input frequency 3 (Hz)" }, { "path": "/Ac/ActiveIn/L1/P", "type": "float", "name": "Input power 1 (W)" }, { "path": "/Ac/ActiveIn/L2/P", "type": "float", "name": "Input power 2 (W)" }, { "path": "/Ac/ActiveIn/L3/P", "type": "float", "name": "Input power 3 (W)" }, { "path": "/Ac/In/1/CurrentLimit", "type": "float", "name": "Input 1 current limit (A)" }, { "path": "/Ac/In/1/CurrentLimitIsAdjustable", "type": "enum", "name": "Input 1 current limit is adjustable", "enum": { "0": "No", "1": "Yes" } }, { "path": "/Ac/In/2/CurrentLimit", "type": "float", "name": "Input 2 current limit (A)" }, { "path": "/Ac/In/2/CurrentLimitIsAdjustable", "type": "enum", "name": "Input 2 current limit is adjustable", "enum": { "0": "No", "1": "Yes" } }, { "path": "/Ac/Out/L1/V", "type": "float", "name": "Output voltage phase 1 (VAC)" }, { "path": "/Ac/Out/L2/V", "type": "float", "name": "Output voltage phase 2 (VAC)" }, { "path": "/Ac/Out/L3/V", "type": "float", "name": "Output voltage phase 3 (VAC)" }, { "path": "/Ac/Out/L1/I", "type": "float", "name": "Output current phase 1 (A)" }, { "path": "/Ac/Out/L2/I", "type": "float", "name": "Output current phase 2 (A)" }, { "path": "/Ac/Out/L3/I", "type": "float", "name": "Output current phase 3 (A)" }, { "path": "/Ac/Out/L1/F", "type": "float", "name": "Output frequency (Hz)" }, { "path": "/Ac/Out/L1/P", "type": "float", "name": "Output power 1 (W)" }, { "path": "/Ac/Out/L2/P", "type": "float", "name": "Output power 2 (W)" }, { "path": "/Ac/Out/L3/P", "type": "float", "name": "Output power 3 (W)" }, { "path": "/Dc/0/Voltage", "type": "float", "name": "Battery voltage (V)" }, { "path": "/Dc/0/Current", "type": "float", "name": "Battery current (A)" }, { "path": "/Ac/NumberOfPhases", "type": "float", "name": "Phase count" }, { "path": "/Ac/ActiveIn/ActiveInput", "type": "enum", "name": "Active input", "enum": { "0": "AC Input 1", "1": "AC Input 2", "240": "Disconnected" } }, { "path": "/Soc", "type": "float", "name": "VE.Bus state of charge (%)" }, { "path": "/State", "type": "enum", "name": "VE.Bus state", "enum": { "0": "Off", "1": "Low Power", "2": "Fault", "3": "Bulk", "4": "Absorption", "5": "Float", "6": "Storage", "7": "Equalize", "8": "Passthru", "9": "Inverting", "10": "Power assist", "11": "Power supply", "252": "Bulk protect" } }, { "path": "/VebusError", "type": "enum", "name": "VE.Bus Error", "enum": { "0": "No error", "1": "VE.Bus Error 1: Device is switched off because one of the other phases in the system has switched off", "2": "VE.Bus Error 2: New and old types MK2 are mixed in the system", "3": "VE.Bus Error 3: Not all, or more than, the expected devices were found in the system", "4": "VE.Bus Error 4: No other device whatsoever detected", "5": "VE.Bus Error 5: Overvoltage on AC-out", "6": "VE.Bus Error 6: Error in DDC Program", "7": "VE.Bus BMS connected, which requires an Assistant, but no assistant found", "10": "VE.Bus Error 10: System time synchronisation problem occurred", "11": "VE.Bus Error 11", "14": "VE.Bus Error 14: Device cannot transmit data", "16": "VE.Bus Error 16: Dongle missing", "17": "VE.Bus Error 17: One of the devices assumed master status because the original master failed", "18": "VE.Bus Error 18: AC Overvoltage on the output of a slave has occurred while already switched off", "22": "VE.Bus Error 22: This device cannot function as slave", "24": "VE.Bus Error 24: Switch-over system protection initiated", "25": "VE.Bus Error 25: Firmware incompatibility. The firmware of one of the connected device is not sufficiently up to date to operate in conjunction with this device", "26": "VE.Bus Error 26: Internal error" } }, { "path": "/Alarms/HighTemperature", "type": "enum", "name": "Temperature", "enum": { "0": "Ok", "1": "Warning", "2": "Alarm" } }, { "path": "/Alarms/LowBattery", "type": "enum", "name": "Low battery", "enum": { "0": "Ok", "1": "Warning", "2": "Alarm" } }, { "path": "/Alarms/Overload", "type": "e [... cut ...] ": "mqtt-broker", "z": "", "name": "", "broker": "192.168.10.167", "port": "1883", "clientid": "", "usetls": false, "compatmode": false, "keepalive": "60", "cleansession": true, "birthTopic": "", "birthQos": "0", "birthPayload": "", "closeTopic": "", "closeQos": "0", "closePayload": "", "willTopic": "", "willQos": "0", "willPayload": "" } ]`

After taking a look at your flow, I think this: [https://github.com/victronenergy/dbus-mqtt](https://github.com/victronenergy/dbus-mqtt) will interest you.

davenrea, 2021-02-23:
Thanks for the link, to far down the road now to make any changes, just need to work out ho to interrogate my solaredge system!!

[image]

riker65, 2021-03-19:
Hi 
I am facing the same issue?

How to delete the flow via commandline?

I am not aware adding a duplicate node....

SergioRius, 2021-03-19:
> Hi
> I am facing the same issue?
> 
> How to delete the flow via commandline?
> 
> I am not aware adding a duplicate node....

First over all, backup your flows file and keep it safe.
Then, when it happened to me for a second time, y just stopped node-red, edited the flows file, searched for victron and carefully deleted the offending node (xml tag).

scottnzuk, 2021-03-22:
I still cant remove this plugin / almost at point of deleting and starting again.. I tried to remove from .config.nodes.json

How can I just factory wipe and reload defaults.

PR #69 (2021-01-13): fix: don't add duplicate victron-client node when loaded
Fixes #66