victronenergy/node-red-contrib-victron #75: Node to read the "Active SoC Limit" (not the minimum SOC)

IoTPlay, 2021-04-30:
Hi, does a node exist to read the Active State of Charge limit?

[image]

mpvader, 2021-08-20:
The solution for this is to add it to the ESS-output node. For which I think we already have an issue or todo item; assigning to dirkjan.

dirkjanfaber, 2021-08-25:
Within the Venus gui, the _Active SOC limit_ is only shown if the Mode is set to _Optimized (with BatteryLife)_. 

The question is if we want to replicate the same logic here or just always have _Active SOC limit_ as a selectable input and let the user decide if it contains actual useful information.

dirkjanfaber, 2021-08-25:
On the `faberd/bms-75` branch, I've added a selectable property for the path object, allowing to set the option to be selectable or not. This results in an unselectable grayed out option to be shown if the Mode is set to something else than _Optimized (with BatteryLife)_:  
[image]

mpvader, 2021-08-25:
Nice! So apparently node-red itself already supports this for properties? Or is all of that configuration window UI defined in our own code?

dirkjanfaber, 2021-08-25:
It is all defined in our own code. (The _buildEditPanel_ function in `src/nodes/victron-nodes.html` to be precise)

closed 2021-09-10 by dirkjanfaber