victronenergy/node-red-contrib-victron #77: Duplicate Flow Import after relaunching Browser

hybridlabor, 2021-06-03:
Hello, I've read about the issue on importing duplicate nodes from victron-client. This happens when relaunching the Browser and requires me to login to the shell and erase the @victronenergy/node-red-contrib-victron manually inside the .node-red userdirectory, relaunch node-red, reinstall the package. 
Running the latest version Node-Red 1.3.5 - as I've read this issue was encountered in 1.3.2 and supposedly fixed but it does not look like it. Any advice on automating the fix / and/or disabling the Warning on duplicate nodes so that the Editor is functioning normally?
Thanks in advance

hybridlabor, 2021-06-15:
Another Issue discovered now, is when going through the trouble of removing / re-adding victronenergy node package, the existing flows I've created will drop all victron elements on the first tab (not where they were originally) and it requires me to re-attach every element.

dirkjanfaber, 2021-08-22:
Please update your palette to load the latest released version (1.4.1). This version contains the fix (which was already done months ago). Until last week, unfortunately enough, this version was not available for download from the palette (it still found version 1.3.2.) 
That has been resolved and updating from the palette should now find version 1.4.1 and fix both of your issues. If not, please open a new issue. I'll close this one for now.

closed 2021-08-22 by dirkjanfaber