victronenergy/node-red-contrib-victron #78: RFE for Additional nodes

fredsparkle, 2021-06-29:
I could make use of the following node types:

1. Node or mode to get network power production across all solar charge controllers.
2. Node to simulate a non-Victron inverter so that it appears in VRM reports.
3. Node to simulate a non-Victron AC charger or AC input draw from shore power.

mpvader, 2021-08-20:
Hi @buddapps , 

For your first one: what we should probably do is update the System node to have more data. Right now it just has two things, power and current. For which it isn't even clear what they are: power of what? and current of what?

[image]

I'll make a new issue  for that.

For 2 and 3, see this one:
https://github.com/victronenergy/node-red-contrib-victron/issues/82

I'll close this one, to keep the list cleaned up.

mpvader, 2021-08-20:
Issue that includes getting the total for all charge controllers is here: https://github.com/victronenergy/node-red-contrib-victron/issues/84

closed 2021-08-20 by mpvader