victronenergy/node-red-contrib-victron #79: solar charger node needs updating for mppt rs

mpvader, 2021-08-14:
The mppt rs has multiple trackers, and such multiple paths with PV information.

https://github.com/victronenergy/venus-private/issues/135

dirkjanfaber, 2021-09-10:
Merged via https://github.com/victronenergy/node-red-contrib-victron/pull/88.

mpvader, 2021-09-10:
Hi @dirkjanfaber : sorry,, but you need to remove /Pv/I and its similar paths, since we'll also be removing those from the D-Bus as well.

The PV array related paths to support in node-red are:
'/Pv/0/V': {'code': 'PVV0', 'whenToLog': 'onIntervalAlways', 'precision': 2},
'/Pv/1/V': {'code': 'PVV1', 'whenToLog': 'onIntervalAlways', 'precision': 2},
'/Pv/2/V': {'code': 'PVV2', 'whenToLog': 'onIntervalAlways', 'precision': 2},
'/Pv/3/V': {'code': 'PVV3', 'whenToLog': 'onIntervalAlways', 'precision': 2},
'/Pv/0/P': {'code': 'PVP0', 'whenToLog': 'onIntervalAlways', 'precision': 0},
'/Pv/1/P': {'code': 'PVP1', 'whenToLog': 'onIntervalAlways', 'precision': 0},
'/Pv/2/P': {'code': 'PVP2', 'whenToLog': 'onIntervalAlways', 'precision': 0},
'/Pv/3/P': {'code': 'PVP3', 'whenToLog': 'onIntervalAlways', 'precision': 0},

and:
`'/Pv/V': {'code': 'PVV', 'whenToLog': 'onIntervalAlways', 'precision': 2},`

and:
`'/Yield/Power': {'code': 'PVP', 'whenToLog': 'onIntervalAlways', 'precision': 0},`

I've added it also to the docs: https://github.com/victronenergy/venus/wiki/dbus#solar-chargers

And where did you learn about `"2": "Error",` being a valid option for /Mode? Its new to me that thats possible, so want to make sure that actually exists.

mpvader, 2021-09-10:
ps. above change is in progress in Venus OS at the moment. Thiemo finished it for vedirect-interface yesterday. And installed that manually on this system: https://vrm.victronenergy.com/installation/55013/dashboard. So if you want to see it, login there.

dirkjanfaber, 2021-09-10:
Adjusted the `services.json` file to accommodate for the d-bus changes.

closed 2021-09-10 by dirkjanfaber