victronenergy/node-red-contrib-victron #82: Add node that allows making a new service on the D-Bus

mpvader, 2021-08-20:
Right now, most nodes we have allow reading data. And a few nodes (ESS, and some more) can control systems.

But there is no node yet that actually puts a new service on the D-Bus. A use case would be reading data from a BMS via Node-red, for example from a serial port or something else; and then wanting to publish that onto the D-Bus for further usage within the Victron system as well as onto VRM. Note that if this is a good idea (doing this for a BMS) is debatable, but perhaps an already easier example is a temperature sensor; or data from an AC Load purely to be able to monitor it together with the rest of the system.

Some issues asking for this are:
https://github.com/victronenergy/node-red-contrib-victron/issues/73
https://github.com/victronenergy/node-red-contrib-victron/issues/78

Note that on our (Victron) todo list, making this is not planned anytime soon. Finishing whats already there, making it official as well as documented has priority. But ofcourse as its all open source, anyone else wanting to do this is very welcome to; and please reach out for that either in this issue or here: https://community.victronenergy.com/spaces/31/mods.html

dirkjanfaber, 2024-11-08:
Resolving this as (partially) solved in version 1.6.0. 

New virtual devices will get added in due time.

closed 2024-11-08 by dirkjanfaber

kevstone122, 2025-09-17:
Hello together,

i have exactly the same problem / idea. How can i add some extra paths for some devices to be later visible in the cerbo gx screen. I have the existing virtual device, but i want to add some additional status value. I have to create some "custom path" to the dBus. How is this possible?

Thank you?