victronenergy/node-red-contrib-victron #84: Improve system node

mpvader, 2021-08-20:
Currently it has only two paths, which is too little; and they are vague in their naming:

[image]

I think they're equal to /Dc/System/Power, which would mean they are whats the "DC System" box on the GX UI.

The purpose of the system node is to have all system totals. And btw not to have ESS features, even though those are (partially) also on the D-Bus under com.victronenergy.system. For ESS, there are ESS nodes.

Paths that should be added to the system node are:

/Dc/Pv/Current  -> Total DC-Coupled-PV Current Current
/Dc/Pv/Power -> Total DC-Coupled-PV Power

/Dc/Battery -> Most of the paths in there as well. Advantage of this is when people make a small flow using battery data, then when they use this it will work on other systems where the battery monitor is connected to a different port as well.

/Ac -> Most of the paths there, to have total of AC-Coupled data for example; and information of the Active AC Input.

mpvader, 2021-08-20:
Note that there is already a pull request that will take care of part or all of the above:
https://github.com/victronenergy/node-red-contrib-victron/pull/61

Next step: check that; and see if more work needs doing or if just including those changes is enough.

dirkjanfaber, 2021-09-10:
Closing this issue as the pull request has been merged.

closed 2021-09-10 by dirkjanfaber