victronenergy/node-red-contrib-victron #91: solar-charger control node can't find a solarcharger service

mpvader, 2021-09-11:
while the output node can find it. This is when running the demo mode; perhaps the issue in there.

the screenshot shows the error, as well as the (connected) output node:
[image]

dirkjanfaber, 2021-09-13:
I can confirm that this issue is there in demo mode, but not with an actual solar charger. The `Mode` path does not exist in `com.victronenergy.solarcharger.tty02` when in demo mode. 

So the cause is out of the scope of node-red-contrib-victron. This issue can be closed.

For gaining my own knowledge. So which scope is it in? The GUI only sets the dbus path `/Settings /Gui/DemoMode` of `com.victronenergy.settings` to a value. But it also looks like it starts playing the recording (from `/opt/victronenergy/dbus-recorder/`). WIthin that directory the file `solarcharger.dat` lacks the '/Mode' path.

closed 2021-09-13 by dirkjanfaber

mpvader, 2021-09-13:
its in no scope @dirkjanfaber , its not a problem that there is no `/Mode` path in the demos.