victronenergy/node-red-contrib-victron #99: Allow controlling ac-out2

SergioRius, 2021-10-01:
Allow controlling the ac-out2 online status to allow scheduling and advanced conditions.
Assistants overriding.

mpvader, 2021-10-01:
that would be nice, but not something we're looking at soon since it involves firmware changes in the multis and quattros.

(same answer as the other one)

SergioRius, 2021-10-01:
I thought that was the main reason for implementing node-red.

mpvader, 2021-10-01:
Controlling things like this you mean? Yes it is, but this particular relay in the multis and quattros is unfortunately not something thats easily made controllable.

we can’t put all our resources on things like this, I hope you understand.

As a solution, wire a power-relay to the signal-relay in the gx device.

SergioRius, 2021-10-01:
I understand.

But if I'm allowed, we have to find some solution to the problem of not having a true zero feed in ESS. Wherever you receive the requirement from the power company, you have to migrate to VS.
But it's cumbersome and you lose so many equipment capabilities.
It's a pity.

dirkjanfaber, 2023-02-03:
Closing this issue as this has to do wait for firmware changes in the multis and quattros and not with the Node-RED nodes.

closed 2023-02-03 by dirkjanfaber

pbawesome, 2024-10-04:
Can we please make this a thing? Having only 2 relays on the cerbo is very limiting. One is taken up by a generator usually, and the other is for my AC, I want another but don't want to go hacking up remote gpio stuff.