victronenergy/node-red-contrib-victron #106: When grid drops, System grid (w) node sends null instead of zero.

SergioRius, 2021-10-31:
[image]

In this case provokes a crash in the following nodes.
Just curious if that's expected.
Could be easily (not straightforward) corrected in after process.

dirkjanfaber, 2021-11-01:
Inserting a function node, containing:
```
if (! msg.payload) {
    msg.payload = 0;
}

return msg;
```
will replace the null value with a zero. 

But I am going to try to get some more background information on this.

dirkjanfaber, 2021-11-01:
We want to differentiate between "device is not providing data" (`null` payload) and "device is measuring 0W" (`0` value). So this is planned behavior. You can expect the `null` value to remain possible in future updates.

closed 2021-11-01 by dirkjanfaber

SergioRius, 2021-11-01:
> Inserting a function node, containing:

Yes, or a change node to follow with:
[image]

> We want to differentiate between "device is not providing data" (null payload) and "device is measuring 0W"

OK. Just for clarification, I'm referring here to the inner AC measurements of a Multiplus2 when it "AC Ignores" the grid. No external meter involved.