victronenergy/node-red-contrib-victron #107: Raspberry Pi Zero 2 W / SignalK - No connection to VE.direct

mapeje, 2021-11-05:
I am running an Openplotter installation with SignalK Node red plugin. When installing node-red-contrib-victron, it cannot connect to my MPPT and VSS. They are however visible in SignalK by the VE.Direct to Signal K plugin.

dirkjanfaber, 2021-11-05:
Can you provide a little more detail on how it is connected and configured? Do you mean you are using https://github.com/SignalK/node-red-embedded ?  And the MPPT is connected via a VE.Direct to USB cable from the Pi? (https://www.victronenergy.com/accessories/ve-direct-to-usb-interface) Is the Pi running Venus OS or Raspbian? 
A few screenshots of what is and what isn't working might also be useful. 
And the abbreviation VSS does not directly ring a bell. Can you explain what that is?

mapeje, 2021-11-05:
OK. I have a RPIZero2W running Openplotter (https://openmarine.net/openplotter) (Raspian). Here I run SignalK (https://signalk.org/)  and from here I have installed Node-red (https://github.com/SignalK/node-red-embedded). I have then used palletes to install node-red-contrib-victron. Other nodes work fine. The Ve-direct cables are connected to RPI using Serial TTYUSB1 at  38400bauds.
I get the data from the MPPT and Victron Smart Shunt (VSS) into SignalK, so there is a working connection

[image]
[image]

dirkjanfaber, 2021-11-05:
Thank you. This makes it a lot more clear. 
The nodes of node-red-contrib-victron rely  on the availability of the `dbus` service, which is available on the typical Venus OS installs. The nodes get and set all of the information on the dbus and not from the serial connection. On Venus OS there is another process that interacts between the serial connection and the dbus. The quick fix would be to add an extra Raspberry Pi to your setup that runs Venus OS and connect the VE.Direct dongles to that Pi. Like described on https://www.victronenergy.com/blog/2019/06/13/sailing-with-signal-k/.  And adjust your current Pi to connect via dbus over TCP, as described [here](https://github.com/victronenergy/node-red-contrib-victron#installation-and-usage). 
The other solution is to make serial connection support within node-red-contrib-victron. I'll take a look alt how difficult that is to implement. But that will probably take some time before being finished. Possibly it can be copied from how it is done in Signal K.

mapeje, 2021-11-05:
Thanks for quick reply! I fully understand. I think running VenusOS on a separate Pi would be a good idea, however it is not yet supported on RPI zero 2W. If this can we solved, I could simply run a Venus OS Large install and have both SignalK and Victron VE work on a single pi.
Perhaps another option might be to use modbus or mqtt to get the MPTT/VSS in my Node-Red?
With the support for serial, it would just be amazing!

dirkjanfaber, 2021-11-05:
I think that if we get it to work, the Venus OS on the RPI zero 2W would be the best route. Noticed that you are also started a thread on that on https://community.victronenergy.com/questions/106779/victron-os-on-new-raspberry-pi-zero-2-w.html. 
At the moment I don't have a RPI zero 2W (yet), so I ordered one. 
As said, I'll also look into the serial connection support for node-red-contrib-victron. I would not start looking into extra routes to get this to work for now.

dirkjanfaber, 2023-02-27:
There now is https://flows.nodered.org/node/node-red-contrib-victron-vedirect-usb for connecting via VE.Direct. Also there is https://flows.nodered.org/node/node-red-contrib-victron-modbus for connecting Victron equipment via modbus.
Either one of these 2 can be used for solving this issue. 

Only hurdle that I can think of is if you want to use both SignalK and Node-RED to connect via the same VE.Direct cable. That probably won't (and afaik can't) work.

closed 2023-02-27 by dirkjanfaber