victronenergy/node-red-contrib-victron #108: VE.BUS System Node missing options

shaneyake, 2021-11-06:
Hi,

There are quite a few options missing from the VE.Bus System Node
Main one that I care about is:
Feed DC overvoltage into grid : /Hub4/DoNotFeedInOvervoltage

But others would be nice:
Disable PV inverter : /PvInverter/Disable
Maximum overvoltage feed-in power L1 : /Hub4/L1/MaxFeedInPower
Maximum overvoltage feed-in power L2 : /Hub4/L2/MaxFeedInPower
Maximum overvoltage feed-in power L3: /Hub4/L3/MaxFeedInPower
VE.Bus Reset : /SystemReset

Thank you

shaneyake, 2021-11-10:
Thanks for the fast fix, I will have to wait for next version of Venus Large but will get back to you.

dirkjanfaber, 2021-12-03:
As far as I have tested this myself, this works as intended. Merged this into master. 
Closing this issue. If not ok, please re-open the issue or create a new one.

closed 2021-12-03 by dirkjanfaber