victronenergy/node-red-contrib-victron #130: Cannot control Relay 1 but Relay 2

erichabg, 2022-04-11:
Config:
Cerbo GX FW v2.82-large-30
Multiplus 24/3000/70-16 Typ:2612 FW 494 (I not II !)
BMV-700 Typ: 0x203 FW v3.11
SmartSolar Charger MPPT 250/60 Typ: 0xA068 FW v1.59
BlueSolar Charger MPPT 75/15 FW v1.59 HQ152982IUA

Problem:
I can control Venus Relay 2 of the Cerbo with the Node but not Venus Relay 1.
There is no error message within nodered but no state change of the Relay and the register.
But I can control both relays with direct modbus access - node via Unit ID 100 and Register 806/807.
So I would expect that Relay 1 works even within nodered as with Relay 2.
Also the relay of the BMV and also the MPPT 250 cannot be controled (neither with node-red nor with modbus).
In the Spec I see that this register is not writable, but then why is it offered to be switched in the node?

dirkjanfaber, 2022-04-11:
Just to be sure. Did you set the Relay 1 to be switchable manual? If so, switching the relays on a Cerbo ought to work without problems. Just tested it on a Cerbo on my desk.
[image]

dirkjanfaber, 2022-04-11:
I also tested it on a BMV. You need to set it to "Remote" via VictronConnect first. After that, you'll be able to switch it with Node-RED. 

[image]

dirkjanfaber, 2022-04-11:
For the MPPT there is also an option, though a bit hidden right now. You could set the relay to load output in VictronConnect. That would reproduce what the virtual load output does. Then you set the virtual load output to always ON. After which it should be possible to control it via Node-RED.
I have not tested this myself as I haven't got a comparable MPPT around right now.

erichabg, 2022-04-11:
> Just to be sure. Did you set the Relay 1 to be switchable manual? If so, switching the relays on a Cerbo ought to work without problems. Just tested it on a Cerbo on my desk. 

Yes, both are set to "Manual" on my side.

erichabg, 2022-04-11:
> I also tested it on a BMV. You need to set it to "Remote" via VictronConnect first. After that, you'll be able to switch it with Node-RED.
> 
> [image]

OK, I think this was the reason. Thank you for the info!

erichabg, 2022-04-11:
> 

> For the MPPT there is also an option, though a bit hidden right now. You could set the relay to load output in VictronConnect. That would reproduce what the virtual load output does. Then you set [...]

I have tried it. Was not so a good idea.  Now the relay hangs in the on mode and I cannot set it to off anymore. Also setting back to "low battery" mode doesn't work anymore....

dirkjanfaber, 2022-04-11:
> I have tried it. Was not so a good idea. ;-) Now the relay hangs in the on mode and I cannot set it to off anymore. Also setting back to "low battery" mode as (before) doesn't work anymore. I had to disconnect the PV modules and the battery to get it working again. I now prefer to keep my hands off this device. ;-) The only important thing for me is to be able to switch the relays on the Cerbo.

I am sorry for the unexpected result. I'll be able to test it on Wednesday and hopeful that I'll come with a working solution.

erichabg, 2022-04-11:
The error with the hanging relay of the MPPT 250 has absolutely nothing to do with the node-red node. This has more to do with VEConfig or the FW of the MPPT. And the problem with the non-switchable relay 1 is not so serious for me, because I have the bypass via the modbus node. But it would be interesting to know why this does not work with my installation. Is there any way for me to debug this?
How does the node access the relays? Also via modbus and Register 806/807?

dirkjanfaber, 2022-04-12:
> How does the node access the relays? Also via modbus and Register 806/807?

The node-RED nodes access all settings via the `dbus`. See https://github.com/victronenergy/venus/wiki/dbus and https://github.com/victronenergy/venus/wiki/dbus-api for more information about this.

dirkjanfaber, 2022-04-12:
> The error with the hanging relay of the MPPT 250 has absolutely nothing to do with the node-red node. This has more to do with VEConfig or the FW of the MPPT. And the problem with the non-switchable [...]

Can you make sure the minimum closed time of the relay is set to 0? You can see that in the relay setting page on VictronConnect.

erichabg, 2022-04-12:
> Can you make sure the minimum closed time of the relay is set to 0? You can see that in the relay setting page on VictronConnect.

The minimum closed time of the MPPT 250 relay is already set to 0m.  For the relays of the Cerbo I find no setting for that.

dirkjanfaber, 2022-04-13:
Getting back to the Cerbo and your first relay not functioning from Node-RED. I just installed the _2.82-large-30_ on my Cerbo to make sure that it is not a firmware version thing. On my system I can control both relays without problems. So there must be some setting different on yours.  
What I can think of is that you can test it two ways: via the gui or from the commandline.

### gui
Via the gui you should be able to switch the relay by setting the relay to Manual and activating/deactivating the line beneath.
[image]
The gui is accessible from a GX Touch device or on the lan via http://venus.local/

### cli
If you've used the commandline before (Check [enable ssh and log in ](https://www.victronenergy.com/live/ccgx:root_access#enable_sshd_and_log_in) for more info), you can also test it via the commandline. If so, can you login with ssh and paste the output of the command:  
```dbus -y com.victronenergy.settings /Settings/Relay/Function GetValue```
Output should be `2`. 

You should also be able to switch the relays from the commandline:  
```
dbus -y com.victronenergy.system /Relay/0/State SetValue %0
dbus -y com.victronenergy.system /Relay/0/State SetValue %1
```
Where the `%0` switches the relay off and the the `%1` switches it on. If that works, it should also work from Node-RED. On the Node-RED side I can only think of that you send something different than a number '0' and '1' to the relays. But that should give some debugging output. 

Under the hood the commandline, the gui and Node-RED do exactly the same: modify the dbus setting upon which the relay reacts.

erichabg, 2022-04-13:
> ### gui
> Via the gui you should be able to switch the relay by setting the relay to Manual and activating/deactivating the line beneath. [image] The gui is accessible from a GX Touch device or on the lan via http://venus.local/
> 
Works without any  problem

> ### cli
> You should also be able to switch the relays from the commandline:
> 
> ```
> dbus -y com.victronenergy.system /Relay/0/State SetValue %0
> dbus -y com.victronenergy.system /Relay/0/State SetValue %1
> ```
> 
> Where the `%0` switches the relay off and the the `%1` switches it on. If that works, it should also work from Node-RED. On the Node-RED side I can only think of that you send something different than a number '0' and '1' to the relays. But that should give some debugging output.

Works without any problem
 \# dbus -y com.victronenergy.settings /Settings/Relay/Function GetValue
2
 \# dbus -y com.victronenergy.system /Relay/0/State SetValue %0
0
 \# dbus -y com.victronenergy.system /Relay/0/State SetValue %1
0

> 
> Under the hood the commandline, the gui and Node-RED do exactly the same: modify the dbus setting upon which the relay reacts.

But node red still not working...

I add my  test Flow. I hope there are no keys and passwords included?

erichabg, 2022-04-13:
[testflow.zip](https://github.com/victronenergy/node-red-contrib-victron/files/8480733/testflow.zip)

dirkjanfaber, 2022-04-13:
> [testflow.zip](https://github.com/victronenergy/node-red-contrib-victron/files/8480733/testflow.zip)

I can duplicate your problem with this flow. I see where the problem lies, just not how it got there. In the json I see:
```
        "serviceObj": {
            "service": "com.victronenergy.system",
            "name": "Venus device",
            "paths": [
                {
                    "path": "/Relay/0/State",
                    "type": "enum",
                    "name": "Venus relay 1 state",
                    "enum": {
                        "0": "Open",
                        "1": "Closed"
                    },
                    "writable": true,
                    "disabled": true
                },
                {
                    "path": "/Relay/1/State",
                    "type": "enum",
                    "name": "Venus relay 2 state",
                    "enum": {
                        "0": "Open",
                        "1": "Closed"
                    },
                    "writable": true
                }
            ]
        },
```
The `"disabled": true` seems to be the cause of the relay not working. Can you try to import the following flow? 
```
[{"id":"a65d7075155db693","type":"inject","z":"2a0477896bc231c9","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"0","payloadType":"num","x":470,"y":240,"wires":[["b8e80a063cf57530"]]},{"id":"f736129d76364948","type":"inject","z":"2a0477896bc231c9","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"1","payloadType":"num","x":470,"y":280,"wires":[["b8e80a063cf57530"]]},{"id":"b8e80a063cf57530","type":"victron-output-relay","z":"2a0477896bc231c9","service":"com.victronenergy.system","path":"/Relay/0/State","serviceObj":{"service":"com.victronenergy.system","name":"Venus device","paths":[{"path":"/Relay/0/State","type":"enum","name":"Venus relay 1 state","enum":{"0":"Open","1":"Closed"},"writable":true},{"path":"/Relay/1/State","type":"enum","name":"Venus relay 2 state","enum":{"0":"Open","1":"Closed"},"writable":true}]},"pathObj":{"path":"/Relay/0/State","type":"enum","name":"Venus relay 1 state","enum":{"0":"Open","1":"Closed"},"writable":true},"initial":"","name":"","x":740,"y":240,"wires":[]}]
```

Changing the "Measurement" from 'Venus relay 1 state' into 'Venus relay 2 state', deploy and changing it back to 1 and deploying again ought to function on your flow as well. I can't find/reproduce where the `"disabled": true` came from.

Btw getting the Relay on the MPPT to work can currently only be done via a hack and not easily via Node-RED. Via VictronConnect you can set the relay mode to "Load output" (as suggested before). After this change, you can _in VictronConnect_ switch the relay by putting the Load output in "Always Off" and "Always On" mode. Setting the load output via dbus is currently not possible (only by changing the values within Vregs), making it also practically impossible to change from Node-RED. I'll try to get the functionality in a new firmware update for the MPPT, but that might take a while. 

Btw 2: passwords are not stored in the `flows.json` file. They usually reside in the `.config.runtime.json` file.

erichabg, 2022-04-13:
Wonderfull!  With your Flow it works!
Now I have delete the Relay-node for Relay 1 from my flow and added it new and - It works! 

May be the problem was caused by copying a Relay-node with setting "BMV" or "MPPT" and change it afterwards  to Venus Relay 1?  I don't know if I generated it that way, but it could be. 
Or the setting for the relay in the console was not "Manual" in the moment I created  the flow and therefore the "disabled" was right in this moment? 
Please don't be angry if I don't try this now by myself, but since I have something connected to the relay (the TSense input of the Multiplus to disable Charge). I do not want to switch here constantly.
The problem with the MPPT relay is not important for me. I have only tried to switch this, because the problem with Venus Relay1 existed.
Thank You for your wonderfull support!  Great!

BTW! It would be wonderfull if it would be possible to switch "Ignore AC" and "Charge enabled/disabled" directly with node-red/dbus etc.  These are the two things I switch by means of the two relays. (Relay 1 closes the TSense input of the Multiplus, Relay 2 closes the Aux input and the corresponding assistant provide the desired function. Not Kiss like ;-) ). I know that would mean an extension of the firmware.

dirkjanfaber, 2022-06-13:
I'll close this issue as it seems to be resolved by now. For the other suggestions please open a new issue (if needed), in order to keep things seperated.

closed 2022-06-13 by dirkjanfaber