victronenergy/node-red-contrib-victron #131: The deleting of nodes is still not working 100% as expected and some nodes will not allow deletion

dirkjanfaber, 2022-04-13:
Via https://github.com/victronenergy/venus/issues/378#issuecomment-1097748566 and https://github.com/victronenergy/venus/issues/378#issuecomment-1098055486

dirkjanfaber, 2022-04-13:
Tested to reproduce this on _v2.82-large-30_, but failed so far:
```
2022-04-13 13:39:23.706447500 13 Apr 15:39:23 - [info] Uninstalling module: node-red-contrib-bigtimer
2022-04-13 13:40:59.944497500 13 Apr 15:40:59 - [info] Uninstalled module: node-red-contrib-bigtimer
2022-04-13 13:40:59.949012500 13 Apr 15:40:59 - [info] Removed node types:
2022-04-13 13:40:59.949726500 13 Apr 15:40:59 - [info]  - node-red-contrib-bigtimer:bigtimer
```

xsilvergs548, 2022-04-13:
I also have difficulty un-installing nodes, please see attached images. 
Many thanks for all the great work.
[image]
[image]
[image]

houser42, 2022-04-13:
node-red-contrib-bigtimer
2.8.1

node-red-contrib-dsm
0.14.1

node-red-contrib-influxdb
0.6.1

node-red-contrib-looptimer
0.0.8

node-red-contrib-solar-power-forecast-plus
2.0.6

Here is my complete list of news that can not be uninstalled if that helps.

node-red-contrib-timerswitch
1.4.0

node-red-contrib-ui-digital-display
1.0.3

node-red-contrib-xterm
1.0.1

dirkjanfaber, 2022-04-13:
This could very well be a remnant from an older version. We try to be backward compatible and not to break things when updating, but sometimes things get overlooked and this might be such a thing. If I install a node via the palette and remove it with `rm` on the Cerbo, Node-Red still thinks they exist (and they do in `package.json`). Not too sure if there is a command to have npm re-investigate the existing packages, but editing the file by hand is a possibility.

dirkjanfaber, 2022-05-12:
I'll close this issue as it seems to be a remnant from older upgrades. Not much to do about it apart from removing the nodes from `package.json`.