victronenergy/node-red-contrib-victron #132: Best practice for CPU and process monitoring of a GX device?

houser42, 2022-04-14:
Hello,

I would appreciate a flow up on the kind advice on the Venus OS large thread if possible?
Some links were given there and are linked below, but perhaps there are others that can be linked here for all to use?
[This one](https://github.com/victronenergy/node-red-contrib-victron/wiki/Example-Flows#monitor-disk-usage) works perfectly to monitor disk space
[This one](https://github.com/bartbutenaers/node-red-contrib-process-resources) looks very useful, but does not seem to connect to a Cerbo GX. Can you suggest a modification to do so?
[This NR flow ]([{"id":"384924bd.c7b6dc","type":"exec","z":"106a5b82.ef95a4","command":"sudo systemctl restart nrlive","addpay":true,"append":"","useSpawn":"","timer":"","name":"","x":873,"y":617,"wires":[["c8f2f6f1.370d08"],["c8f2f6f1.370d08"],["c8f2f6f1.370d08"]]},{"id":"5c9d8978.a36278","type":"inject","z":"106a5b82.ef95a4","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":674,"y":617,"wires":[["384924bd.c7b6dc"]]},{"id":"c8f2f6f1.370d08","type":"debug","z":"106a5b82.ef95a4","name":"","active":true,"console":"false","complete":"false","x":1143,"y":617,"wires":[]},{"id":"46129a4f.b9ed64","type":"comment","z":"106a5b82.ef95a4","name":"** DANGER!: Restart the Node-Red service **","info":"","x":770,"y":580,"wires":[]},{"id":"85941f67.76613","type":"inject","z":"106a5b82.ef95a4","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":673,"y":708,"wires":[["f5f9bc5a.5ab3f"]]},{"id":"f5f9bc5a.5ab3f","type":"exec","z":"106a5b82.ef95a4","command":"sudo reboot now","addpay":false,"append":"","useSpawn":"","name":"","x":842,"y":708,"wires":[["e5e0f45f.0287f8"],["e5e0f45f.0287f8"],["e5e0f45f.0287f8"]]},{"id":"e5e0f45f.0287f8","type":"debug","z":"106a5b82.ef95a4","name":"","active":false,"console":"false","complete":"false","x":1142,"y":708,"wires":[]},{"id":"5714af73.78418","type":"comment","z":"106a5b82.ef95a4","name":"** DANGER! Reboots the Pi running Node-Red, InfluxDB and MQTT **","info":"","x":847,"y":672,"wires":[]},{"id":"7e632c5f.0a5944","type":"comment","z":"106a5b82.ef95a4","name":"TODO: Remove service name and pull from settings instead","info":"","x":1160,"y":580,"wires":[]}])from here is something I really need: to be able to restart the node red instance from within itself.
Can you suggest how to modify it to work on a Cerbo GX also?

dirkjanfaber, 2022-04-14:
> 1. to be able to restart the node red instance from within itself via automation and a dashboard switch.

You'll  take the chance of shooting yourself in the foot with this. Node-RED can stop itself, but might not be able to recover from it. The complete command is `cd /service && svc -i node-red`, which seems to work. 

> 2. Also to reboot the Cerbo GX device completely.
Can you suggest how to modify it to work safely and robustly on a Cerbo GX also?

You can leave out the `sudo` and the `now`, leaving only `reboot` and it should work. We don't ship with sudo on Venus right now.

That being said, we are in the process of making Node-RED part of the normal image. One of the planned changes is to run Node-RED as a normal user instead of as the `root` user as it is on the current image. Once that change has been made, the above will probably break. 

The adjusted flow:  
```
[{"id":"384924bd.c7b6dc","type":"exec","z":"a7db6b9964babd4d","command":"cd /service && svc -i node-red","addpay":"payload","append":"","useSpawn":"false","timer":"","winHide":false,"name":"","x":370,"y":180,"wires":[["c8f2f6f1.370d08"],["c8f2f6f1.370d08"],["c8f2f6f1.370d08"]]},{"id":"5c9d8978.a36278","type":"inject","z":"a7db6b9964babd4d","name":"","repeat":"","crontab":"","once":false,"topic":"","payload":"","payloadType":"str","x":161,"y":180,"wires":[["384924bd.c7b6dc"]]},{"id":"c8f2f6f1.370d08","type":"debug","z":"a7db6b9964babd4d","name":"","active":true,"console":"false","complete":"false","x":630,"y":180,"wires":[]},{"id":"46129a4f.b9ed64","type":"comment","z":"a7db6b9964babd4d","name":"** DANGER!: Restart the Node-Red service **","info":"","x":257,"y":143,"wires":[]},{"id":"85941f67.76613","type":"inject","z":"a7db6b9964babd4d","name":"","repeat":"","crontab":"","once":false,"topic":"","payload":"","payloadType":"str","x":160,"y":271,"wires":[["f5f9bc5a.5ab3f"]]},{"id":"f5f9bc5a.5ab3f","type":"exec","z":"a7db6b9964babd4d","command":"reboot","addpay":false,"append":"","useSpawn":"false","timer":"","winHide":false,"name":"","x":289,"y":271,"wires":[["e5e0f45f.0287f8"],["e5e0f45f.0287f8"],["e5e0f45f.0287f8"]]},{"id":"e5e0f45f.0287f8","type":"debug","z":"a7db6b9964babd4d","name":"","active":false,"console":"false","complete":"false","x":629,"y":271,"wires":[]},{"id":"5714af73.78418","type":"comment","z":"a7db6b9964babd4d","name":"** DANGER! Reboots the Pi running Node-Red, InfluxDB and MQTT **","info":"","x":334,"y":235,"wires":[]},{"id":"7e632c5f.0a5944","type":"comment","z":"a7db6b9964babd4d","name":"TODO: Remove service name and pull from settings instead","info":"","x":647,"y":143,"wires":[]}]
```

Getting `node-red-contrib-process-resources` to function takes a bit more work. I see that the cause is that it uses a `ps -A -o ppid,pid` under the hood to get the process information. As Venus OS ships with `ps` from BusyBox, those parameters don't exist. It can probably be fixed by getting the Information from the `/proc` filesystem. I need to think about this solution, and get back to it.

houser42, 2022-04-14:
Excellent replies, many thanks @dirkjanfaber 

Will test that updated flow asap..

If it needs to be said, these are wishes for what I feel would certainly be very useful stuff, but I am not 
in a hurry through life. So ,maybe just keep these things in mind for your processes if it makes sense at some  point.

About the "shooting in the foot" stuff. I certainly understand that.
My thing is that I now have a really useful Node Red setup on our Cerbo GX but it gets a bit bogged down
over time and needs restarting to be snappy again. I am trying to figure out what does this, as I don't think my setup
is that big. The most process intense is probably the Node Red Dashboard with more than one user accessing it.
But until I can figure it out and maybe trim my system, the NR setup probably needs automated restarts for now..
Many thanks again!

mpvader, 2022-04-14:
Perhaps good to know that the Cerbo will also reboot itself when it gets too bogged down.

houser42, 2022-04-19:
I did a lot of troubleshooting on my system over the last few days.
I have come to the conclusion that my issue was the frequency of updating the charts on the dashboard. A fair bit of reduction there by using rate limit nodes before the charts appears to have helped a lot. FWIW, YMMV.

dirkjanfaber, 2022-06-13:
I'll close this issue as it seems to be resolved by now.

closed 2022-06-13 by dirkjanfaber