victronenergy/node-red-contrib-victron #133: [bug?] The VRM app appears broken for Venus OS large/Node Red?

houser42, 2022-04-29:
Hello,
Writing this here, as Venus OS large is not part of the main build yet.
I would assume you are already aware of this, but it seems to me that the last version of the iOS VRM app (2.5.6) 
from April 21 has broken the secure tunnelled links to Node red in the app.

I see no issues with this on Safari on the iPhone or the former version () on other iPhone and also no issues on any webbrowsers or OS that I use. If I am missing something or there is some new setting or procedure that needs to be followed with the new app, let me know please? FWIW. Many thanks.

dirkjanfaber, 2022-04-29:
Thanks for reporting. I wasn't aware of this yet. I assume you mean the VictronConnect app. I'll look into it and contact the developer if I can reproduce the issue.

houser42, 2022-04-29:
Hi @dirkjanfaber 
No I do mean the VRM app.

houser42, 2022-04-29:
Using the "Venus OS Large" and the tapping the node red icons appears broken in the app.
iOS tested only.

dirkjanfaber, 2022-04-29:
Clear. I can reproduce the problem and contact the developer. Thanks for reporting. I'll leave this issue open until it has been fixed.

houser42, 2022-05-09:
Just reporting that this works now on my main system, but the app still opens a "No connection Retry" message. But it now opens the page correctly.

houser42, 2022-06-11:
So I still see this inability to open the node red links from within the app on the handful of systems I maintain that miss this really great and useful feature. 
All installations are running latest betas of Venus OS Large 2.90~12 as I write this.
I assume this will be rolled into the VRM app at some point but just making sure I am not missing something, as it's been like this now for a month. Going the webgui path until further notice so no crisis today either, just checking ;)
Many thanks.

dirkjanfaber, 2022-06-11:
I'd wish it would have been resolved sooner too, but it still on the radar. As far as I know I t only affects iOS, so it should work from android devices as well.

dirkjanfaber, 2022-07-01:
To give an update. The buttons do work again on iOS with the latest VRM app. Only getting back into the VRM app afterwards is still problematic (and being fixed).

houser42, 2022-08-10:
This is all working now as far as I can tell. I should mention that this is also now tested  on iOS 16 beta. FWIW.
Many thanks

closed 2022-08-10 by houser42