victronenergy/node-red-contrib-victron #134: Large online update - big blue NR dashboard icon appears to be missing?

houser42, 2022-05-06:
Hello,
I used the new online update method on a Venus GX

Updated to Venus OS large v2.90~3 on a non-critical system from 2.8.large 25 large beta on a Venus GX
It updated ok apart from one thing: 
The big blue Node Red dashboard icon is missing in the VRM menu. Main NR icon is there. Dash not so much.
It is available  if I add /ui to the url manually, but the big blue dashboard icon appears to be missing. FWIW. 
Would appreciate if there is a way to troubleshoot or repair that feature. Many thanks

houser42, 2022-05-09:
Hello,
Just making sure I am not missing something obvious.
The weekend came and went and there is still no NR dashboard icon here.
I assume you have a lot of work to do and will attend to it when you have time.
Or my system has another issue? Maybe worth updating from a USB stick again?
I updated that system to the latest online version .5 too. No change.

dirkjanfaber, 2022-05-09:
It appears that the first fix did not resolve it for all installations.  I am sorry about that.
Background is that one of the checks vrmlogger does to show the VRM dashboard button, is if the directory `/data/home/root/.node-red/node_modules/node-red-dashboard` exists. With the move of node-RED from running as user `root` to user `nodered`, the directory Node-RED creates changed to `/data/home/nodered/.node-red/node_modules/node-red-dashboard`. The vrmlogger (that tells VRM about the dashboard) is still looking at the old directory. With a new installation and an update, the old directory moves to the new location. In some cases (such as on my system) both the `/data/home/root/...` as the `/data/home/nodered/...` directory existed. Which explains why I overlooked this check.
So what you can do, as a quick fix, is a `mkdir -p /data/home/root/.node-red/node_modules/node-red-dashboard`, which should bring the button back. And remove it again when the .6 version comes out.

houser42, 2022-05-09:
Will do. Appreciate the info very much!
Is there an expected ETA for .6 ? Many thanks.

dirkjanfaber, 2022-05-09:
I am not too sure yet, we'll add it together with some other fixes. Should be within a few days.

mpvader, 2022-05-10:
Hi, you just have to either wait for it to be fixed (v2.90~7 most likely).

or roll back to a previous version where it did wirk for you.

I’m sorry, but this is a beta version, its a known issue on our side and nothing more to do or discuss about it.

houser42, 2022-05-11:
I fully understand. I assumed reporting was useful.  Rolling back does however not work either. Ah well. Of course I’ll wait. No worries.