victronenergy/node-red-contrib-victron #135: Simulating energy meter

tompeel, 2022-06-07:
Not a bug, either a feature request or a plea for help.

Super cool Integrating nodered directly, I have my Meyer readings in mqtt, I see I can use the ESS node and the grid set point to change the set point to keep the meter down but Its hard to automate without another value. Unless I'm mistaken.

I just use my current energy use - (current energy use *2) to get a negative and feed it into the grid setpoint.

This doesn't work for the obvious reason I'm missing some way if knowing how much to tell the MPii to push out while still reading the grid.

There must be a way I just can't figure it out?

Any help much appreciated!

dirkjanfaber, 2022-06-10:
Thank you for the kind words.

As you are not describing a bug or feature request in one of the nodes, I will close this issue and redirect you to  https://community.victronenergy.com/ and close this issue as I believe that you will get better help there. Make sure to describe the setup on your usage situation, as that will give you a direction to the solution. The Victron equipment is used in all kinds of different situations. E.g. systems that are completely off grid and use a BMS for overcoming the nights when there is no solar. In other situations the systems are used as protection for a unreliable grid and overcome outages.

If needed, you may also find it helpful to put a grid or energy meter (https://www.victronenergy.com/live/energy-meters:start) into your system. These grid meters integrate with the Victron Grid Meter node.

closed 2022-06-10 by dirkjanfaber

tompeel, 2022-06-12:
Thanks for the reply, i know this is closed but it took me a while to find it so ill just put a closing reply.

The whole point is to avoid the expense and hassle of getting a supported meter, to get a CT clamp one with ethernet is impossible so far as i can tell..

I haven't changed anything in the ESS setting to allow me to do this but i now have a working node red implementation setting the Grid Setpoint with the numbers coming from the  <(Multi ii node AC power L1 Setpoint + ESS Grid Setpoint) - MySetPoint > seems to be working ok. 

only issue is the frequency of data is 5 seconds so you get a bit of over run on a big load off but i think the inverter is slow to adjust so probably more data doesn't help. 

Ill check the forums! thanks