victronenergy/node-red-contrib-victron #136: Node Red Relay control for Relay 4

Benthegreat, 2022-06-20:


dirkjanfaber, 2022-06-20:
In the `services.json` file all of the relays that can be switched from Node-RED are defined. The usual Victron devices (Cerbo GX, CCGX and Venus GX) have only 2 relays. For the Raspberry Pi with the relay hats I already have increased the number of defined relays to 4 (`/usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/services.json`). 

If you want to add more, you can add extra in the `input-relay` / `system` and `output-relay` / `system` part (around line 3235 and line 5017). You can add 2 extra relays there. Just make sure that the json file remains valid. After changing the file you need to restart Node-RED (`cd /service; svc -d node-red-venus; svc -u node-red-venus`). 

As the file is already really large, I am a bit reluctant to add more than 4 to the mainstream release. But that means that you will need to manually add the 2 extra relays after each update for now.

closed 2022-06-22 by dirkjanfaber

Benthegreat, 2022-06-23:
Thank you for the help. I got 5&6 to work.  Thanks for all the hard work you put in.