victronenergy/node-red-contrib-victron #137: Extend Node-Red generator nodes to work with many more generator models

zorrobyte, 2022-07-13:
I am building an ESP32 module with relays over MQTT to control my RV's Cummins Onan generator, a VERY common generator for RVs.

It would be nice to be able to set Victron Generator Alarm/Error state via Node Red. The reason is that many use the Multiplus charger/inverters that only have one AC input and cannot use Victron Generator AC detection. I am using an automatic transfer switch (which came with the RV) to switch between shore and generator.

Another reason is that I'm monitoring the Multiplus VAC 1 input and turning off the generator if AC input is not detected in a few minutes, so the generator doesn't run and consume all fuel when in an error state.

I am not sure if the stock Victron Generator AC detection/Alarm turns OFF the generator, or how long it waits for the generator to start up before throwing an alarm as it is not documented, and would be nice to know. If so, maybe I can set the AC 1 input to generator and take advantage of this alarm state to stop the generator.

Lastly, it would be nice to have a node to stop an automatic start of the generator. This can be forced in Node Red, but if Venus is not aware that the generator is not running, the generator runtime isn't correct.

dirkjanfaber, 2022-07-14:
Thank you for the long description. As a person who is not using generators too often, I am trying to get all the questions answered, but feel free to ask if something is not clear.  

Writing alarms is not something that we want Node-RED to do, but you can take a look at the `/Alarms/NoGeneratorAtAcIn` path, which is part of the `input-generator` module. If you want to dig deeper into the generator part, you might want to take a look at https://github.com/victronenergy/dbus_generator. 

> I am not sure if the stock Victron Generator AC detection/Alarm turns OFF the generator, or how long it waits for the generator to start up before throwing an alarm as it is not documented, and [...]

That time is set to 5 minutes. 

> Lastly, it would be nice to have a node to stop an automatic start of the generator. This can be forced in Node Red, but if Venus is not aware that the generator is not running, the generator runtime isn't correct.

At the moment the generator dbus paths are still not crystalized yet for Venus 2.90. The `/AutoStartEnabled` path is planned to be added in the near future. After that has been done, it will also be added to the generator node.

dirkjanfaber, 2022-09-12:
I believe this issue can be closed now. The generator node is in place with the [input-generator](https://github.com/victronenergy/node-red-contrib-victron/wiki/Available-nodes#generator) and [output-generator](https://github.com/victronenergy/node-red-contrib-victron/wiki/Available-nodes#gx-generator-control) paths available.

closed 2022-09-12 by dirkjanfaber