victronenergy/node-red-contrib-victron #139: Feature wish/idea- node to send data to VRM

houser42, 2022-07-29:
Hello,
Just a thought, as I have been using this excellent software on several PV installs with Victron gear for a while now and this keeps coming up.

It might be very useful for many users to have the ability to route some data from a few nodes in the Node Red environment into the VRM Advanced widget view. Maybe a "Route to VRM node" that allows ,say 10 instances or so, to keep it manageable?
The advantages are probably obvious and I am sure users would find many ways to use such a node but some examples would be: 
* kW and kWh readings from various sources that are in the NR environment, such as smart plugs for longer safekeeping of energy consumption data.
* a custom widget that has several temperature readings like battery and ambient temperature overlayed.
* tank and energy consumption overlayed in a custom widget.
etc..
Again, just a thought, FWIW.

freakent, 2023-04-25:
The obvious answer is to send those readings to your GX device, the GX device will then forward them on to VRM. You could use my generic driver, [dbus-mqtt-devices](https://github.com/freakent/dbus-mqtt-devices), to register your devices and send the data via node-red.

freakent, 2023-04-25:
Yes, I understood the question, although I don't know exactly what you mean by "advanced" widgets. Include a screen shot so there is no misunderstanding. The point is that the data displayed in VRM comes from the GX device. So if you send the correct data to GX, it should be available for you to use in VRM widgets.

houser42, 2023-04-25:
Ok, sorry,  so there is a function called "custom widget" that could potentially be used to overlay something like
humidity over temperature, two different tanks, several house temps or the like. Data I would want to keep in the VRM for maybe a year at a time.
This data is only available as yet on the NR side of things AFAIK. 
This is not vital in any way but would certainly be very nice to have if it could be done.
Many thanks again for your comments.

freakent, 2023-04-25:
I use an advanced widget to show temperature readings from multiple temperature sensors on one chart. That temperature data is fed into the GX device. 

The issue about more than 6 months worth of data is a different one, if you want to hold data over a longer period of time you might want to look into using Grafan as a dashboard.

houser42, 2023-04-25:
Interesting...
Ok, just so we are on the same page:
Are those temp sensors connected software-wise to the native Venus OS side of it or on the Node Red side?

houser42, 2023-04-25:
Ok, not irrelevant on the systems I maintain. But I think I see what you mean. Thanks again.

FidsK, 2025-01-31:
@freakent 
Not to hijack this thread, I think I fully understand what you are saying.
How do I get my data into the Cerbo and therefore VRM using your script?
To clarify, I have Mopeka Universal sensors but their bluetooth range doesn't reach the Cerbo. However, both the Cerbo and sensors are covered by wifi. I can read the data sent by Mopeka in JSON format using custom software.

houser42, 2025-01-31:
> [@freakent](https://github.com/freakent) Not to hijack this thread, I think I fully understand what you are saying. How do I get my data into the Cerbo and therefore VRM>>>

Have you had a look at the Venus  Large integration of Node Red into a GX device?
That night be a way to pulpit together. Good luck.

closed 2025-01-31 by houser42

freakent, 2025-01-31:
@FidsK if you want to continue this discussion please open a discussion on my repo https://github.com/freakent/dbus-mqtt-devices

FidsK, 2025-01-31:
> You can connect to VRM using the virtual tank sensor node.

Thank you for your suggestion. I now have a virtual device running in Node
Red and VRM nicely. I used the virtual temperature sensor for this test. I
will change to the tank level device later. You can see the Node Red and
VRM screenshots of my setup.
Just one challenge: Any idea how I automate the process of getting the
actual sensor data to the inject node (and therefore to the virtual device
and VRM)? I can extract the data from the sensor periodically but I can't
figure out  how to deliver it to the virtual device. What is the process of
writing custom data to a virtual device? (As a suggestion, consider
allowing users to enter custom data in a virtual device through node red).
I realise this probably goes beyond the scope of your current project but
hopefully, you can help.
Fideri

FidsK, 2025-01-31:
Thanks for the quick suggestion. All virtually done and running. I can
input the sensor data manually and it shows up on VRM, no problem. I'm yet
to figure out how to feed it to Victron automatically.
Fideri

dirkjanfaber, 2025-01-31:
You can write to the virtual device using the custom control node. 
There are some examples you can import in Node-RED. It should look something like this:

[image]

In this example I am using the change node to adjust the `msg.payload` to contain `msg.payload.value` (as it comes from this mqtt broker).

FidsK, 2025-02-01:
Thank you. That helped a lot. Now everything is running flawlessly.
Fideri