victronenergy/node-red-contrib-victron #140: Request: Add DVCC parameters to settings output node

liamhogan, 2022-08-17:
Currently the DVCC parameters are controllable by the Settings Input node, under _'DVCC System Max Charge Current'_ and _'Limit Managed Battery Voltage'_ - but there is no way to read the status of these parameters back from an output node (the logical place being the Settings Output node).

I imagine if these parameters are controllable on the dBus then they should be readable as well. Please correct me if this is the case?

dirkjanfaber, 2022-08-17:
Agreed, that would make sense. And currently that is indeed not yet impossible. I'll add these paths.

liamhogan, 2022-08-17:
Ah, just noticed there is already a readable _Charge Current Limit_ on that `input-ess` node. But yes, no voltage. 

Is there a reason you are specifically associating these params with ESS? There are many non-ESS-specific uses, mine included. They are battery management settings. 
I think the varied node locations (split between System Settings and ESS Settings) as well as varied names are a confusion. 
Just calling them 'DVCC Charge Voltage Limit' and 'DVCC Charge Current Limit' and listing them together in the System Settings nodes would make most sense - this is also how they are named in the main GUI, under DVCC.

dirkjanfaber, 2022-08-17:
AFAIK there is not particular reason that it is like this, other than that it grew like this. Keeping the naming consistent in the GUI, VictronConnect, VRM, Node-RED and others (e.g. dbus-modbustcp) is an ongoing process and sometimes we are dependent on users that point out flaws. 
But that does not mean that we cannot make changes (in the last update the `output-genset` node got removed because the `output-generator` offers the same functionality). 
Usually I take the description from dbus-modbustcp, but I agree that adding "DVCC" as prefix makes it more clear. I also agree that it makes sense to move all of the DVCC and ESS settings to the `input-settings` node.

liamhogan, 2022-08-17:
One of the things I have always appreciated about everything Victron is the openness to listen to the community and users and improve the products and services. It is nice to see that there is a similar thing happening here. Thanks Dirk

closed 2022-08-17 by liamhogan