victronenergy/node-red-contrib-victron #153: Endpoint

dowermyr, 2022-12-06:
Hi!
How do I set up the endpoin?
I get this when trying to add a node:
[image]

dirkjanfaber, 2022-12-06:
You probably don't have a battery monitor (a smartshunt or so) connected to your system. It only shows a selectable dropdown when it finds devices on your system. E.g.:  
[image]

The _system_, _relay_ and _settings_ node should always provide an endpoint.

dowermyr, 2022-12-06:
I have a smart shunt. But I haven't got a prompt for my vrm credentials. So where do I type in this?

dirkjanfaber, 2022-12-06:
In order to make sure I understand your situation correctly, as the given data is not that much. For using the Node-RED battery monitor, you should not need to enter your vrm credentials. In fact that is not needed for any of the nodes.

But let's get your situation clear first. Can you respond to:
-  Is your  smart shunt connected via a VE.Direct cable to your Cerbo GX?
- do you see the data from the smart shunt in the (remote) console? 
- which version of Venus are you running?

Please elaborate if you are using a different setup and how you got everything connected to each other.

dirkjanfaber, 2022-12-06:
That is one of the options. Next best is probably to use dbus over tcp as described here: https://github.com/victronenergy/node-red-contrib-victron#installation-and-usage. Basically adding a few entries to `/etc/dbus-1/system.conf` on the Raspberry pi and adding `export NODE_RED_DBUS_ADDRESS=192.168.1.1:78` to the home assistant system before startup of Node-RED.

Alternative is not to use the node-red-contrib-victron nodes at all and use mqtt or modbus. The first one has an example here: https://github.com/victronenergy/node-red-contrib-victron/wiki/Example-Flows#using-mqtt-to-monitor-disk-space The other one has, as far as I know, no example yet. I'll add that soon.

dirkjanfaber, 2022-12-07:
If you have more questions or need extra support, please go to https://community.victronenergy.com/ and ask it there. As this is not an issue with node-red-contrib-victron, I'll close this one.

closed 2022-12-07 by dirkjanfaber