victronenergy/node-red-contrib-victron #160: [BUG] VE.Bus System node - Switch Position

mr-manuel, 2023-02-18:
**Describe the bug**
The `VE.Bus System node` with the `Switch Position` measurement is not setting the Switch Position anymore for a MultiPlus II with the latest version `1.4.27`. When I revert back to `1.4.26` it is working.

**To Reproduce**
Steps to reproduce the behavior:
1. Insert the `VE.Bus System node`
2. Select the `Switch Position` measurement
3. Send a value to it
4. Nothing happens

**Expected behavior**
The value sent to the node should be set.

**Flow**
````json
[{"id":"92b29fb3827ce9d1","type":"inject","z":"e198f4dd338c7082","name":"Charger only","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"1","payloadType":"num","x":110,"y":580,"wires":[["d9bd204c6eb95566"]]},{"id":"00fdc673b2622f3d","type":"inject","z":"e198f4dd338c7082","name":"Inverter only","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"2","payloadType":"num","x":290,"y":580,"wires":[["d9bd204c6eb95566"]]},{"id":"a2f59c8f6d4b77e3","type":"inject","z":"e198f4dd338c7082","name":"On","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"3","payloadType":"num","x":450,"y":580,"wires":[["d9bd204c6eb95566"]]},{"id":"a59446de2c1e8d1b","type":"inject","z":"e198f4dd338c7082","name":"Off","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"4","payloadType":"num","x":590,"y":580,"wires":[["d9bd204c6eb95566"]]},{"id":"d9bd204c6eb95566","type":"show-value","z":"e198f4dd338c7082","name":"","path":"","x":750,"y":580,"wires":[["866132b3cb024dfe"]]},{"id":"866132b3cb024dfe","type":"victron-output-vebus","z":"e198f4dd338c7082","service":"com.victronenergy.vebus/275","path":"/Mode","serviceObj":{"service":"com.victronenergy.vebus/275","name":"MultiPlus-II 48/5000/70-50"},"pathObj":{"path":"/Mode","type":"enum","name":"Switch Position","enum":{"1":"Charger Only","2":"Inverter Only","3":"On","4":"Off"},"writable":true},"initial":"","name":"","onlyChanges":true,"x":1030,"y":580,"wires":[]}]
````

**Hardware (please complete the following information):**
 - Hardware: [e.g. Cerbo GX, Ekrano, Raspberry Pi]

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.4.27
 - Venus OS version: 2.93~3

dirkjanfaber, 2023-02-18:
The above commit is a "quick fix". I'll work on a more permanent shortly. So leaving this issue open.

dirkjanfaber, 2023-02-19:
Tested (with a new test flow) and merged with master.

closed 2023-02-19 by dirkjanfaber