victronenergy/node-red-contrib-victron #171: Temporary network connection problems causes permanent failure

fmonera, 2023-04-05:
I have a node-red 3.0.2 installation on an ubuntu server. It connects to a CCGX via dbus and it works without problems. I have some flows that uses the node-red-contrib-victron nodes both to read and update.

When there is a brief connectivity error (even for a few seconds), the Victron nodes stops working. There are no more updates even when the network is working again. There is nothing I can do except restarting the whole node-red to make it work again. Restarting flows doesn't work. 

**To Reproduce**
One way to reproduce the behavior would be:
1. Configure node-red-contrib-victron to connect via DBUS to CCGX
2. Connect for example AC-Consumption to a Debug node so that it updates the data.
3. Disconnect the network cable for a few seconds (node will stop working)
4. Connect the cable again (node will NOT start working again)

**Expected behavior**
A brief connectivity error should not make me restart the whole Node Red to make Victron Nodes work. It should keep retrying until it works again.

**Hardware (please complete the following information):**
 - Hardware: Cerbo GX

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.4.32
 - Venus OS version: 2.93
 - node-red version: 3.0.2

** Additional notes **
Everything is Ethernet and on the same subnet. There are no Wifi involved in this whole system.

fmonera, 2023-04-05:
Additional information:

 - It used to work without problems some months ago. It doesn't work correctly since latest big change you made on the code (¿v1.4.25?), where I had to reconfigure all victron nodes.
 
 - If I connect to CCGX via Modbus TCP instead of node-red-contrib-victron it works without problems and survives any temporary connectivity problem.

dirkjanfaber, 2023-04-07:
I did some tests, but so far am unable to reproduce it. When I reconnect the network connection to the Cerbo, the missed messages come in and new messages keep on coming in.

I suggest using the work-a-round with the modbus tcp nodes (have you seen [node-red-contrib-victron-modbus](https://flows.nodered.org/node/node-red-contrib-victron-modbus)?). I'll close this issue for now and re-open it once I find a way to reproduce it.

closed 2023-04-07 by dirkjanfaber