victronenergy/node-red-contrib-victron #172: [BUG] Gridmeter node - missing L3 Current (A) entry

mojo2012, 2023-04-24:
The grid meter node is not offering the entry L3 Current (A). The current entry is available for L1 and L2 though.

**To Reproduce**
Steps to reproduce the behavior:
1. insert the node
2. look for L3 Current (A)

**Expected behavior**
It is there and works.

**Screenshots**
[image]

**Hardware (please complete the following information):**
 - Multiplus 2 GX 3000

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.4.32
 - Venus OS version: 2.94

dirkjanfaber, 2023-04-25:
That is weird. It already got added in [this](https://github.com/victronenergy/node-red-contrib-victron/commit/7247f974fb6bece3291e1752c4787940b186cb8d) commit. 

So it should be there. Only thing I can think of that can cause it, is that you are checking a node that pre-dates this change. Can you check if you have _L3 Current (A)_ when you drag a new Grid meter node to the canvas?

mojo2012, 2023-04-25:
I  re-added the gridmeter node as suggested - but no change. Is there a way to debug this?

dirkjanfaber, 2023-04-25:
I've installed firmware 2.94 on a Cerbo and checked. The 2.94 firmware comes with version 1.4.23 and not 1.4.32. 
[image]

So, that pre-dates the change. The simplest way is to update to the latest version is to change your Venus firmware settings to use the update feed "Latest release candidate" instead of "Latest release" and then update to _v3.00~36_. 

A less simple solution is to edit the `/usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/services.json` file with and apply the same patch as mentioned before. And then restart Node-RED (or the Multi).

mojo2012, 2023-04-25:
Damnit, you are right, it's 1.4.23 - sorry my mistake. I manually patched the files and now everything's working :-)
Thanks.

closed 2023-04-25 by mojo2012