victronenergy/node-red-contrib-victron #178: [BUG] Venus settings - DVCC Maximum charge voltage (V) missing in ESS Control

1va13n7in, 2023-07-21:
Venus settings | DVCC Maximum charge voltage (V) is available in node ESS
Venus settings - DVCC Maximum charge voltage (V) is missing in node ESS Control

**Expected behavior**
Venus settings - DVCC Maximum charge voltage (V) should be available in node ESS Control

**Screenshots**
attached

**Hardware (please complete the following information):**
 - Hardware: [Cerbo GX]

**Software (please complete the following information):**

[image]
- node-red-contrib-victron version: 20230505191133
 
- Venus OS version: V3.00~39

dirkjanfaber, 2023-08-15:
As DVCC voltage control won't work with all batteries (see https://www.victronenergy.com/media/pg/CCGX/en/dvcc---distributed-voltage-and-current-control.html), I am a bit hesitant to add it to the ESS control node.

dirkjanfaber, 2023-08-29:
But the new input and output node support this. So feel free to use that (if your battery is supported).

closed 2023-08-29 by dirkjanfaber