victronenergy/node-red-contrib-victron #182: [BUG] Enabling "httpNodeAuth" for dashboard breaks Victron nodes

DarkZeros, 2023-08-23:
**Describe the bug**
If we want to expose the node red dashboard to the internet, it is a good idea to enable this feature and only allow password protected acceses.
But this breaks the victron nodes connection with the VRM in the VenusOS

**To Reproduce**
Steps to reproduce the behavior:
1. Enable "httpNodeAuth" in settings-user.js or settings.js
2. restart nodered or VenusOS
3. Open any victron node
4. Will lask for a password, no password seems to work, it just fails:

**Expected behavior**
Works normally

**Screenshots**
[image]
[image]

**Hardware (please complete the following information):**
 - Hardware: VenusOS on Rpi3B

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.4.32
 - Venus OS version: v3.01 (but also failed on v2.xx that is why I updated)

DarkZeros, 2023-08-23:
As I understand this problem is occuring because the Victron UI is requesting https://localhost:1881/victron/services/input-solarcharger and so on addresses to the server, and this requires authentication for httpNodeAuth. Looks like abug to me, since the feature is incompatible with a feature in node red.

dirkjanfaber, 2023-08-23:
Alright, I see a possible overlooked feature here. I'll reopen the issue and check if user authentication can somehow be configured. Mind that the node-red-contrib-victron node don't use the http endpoints while running, it is only used on the edit panel (as other nodes also do).

mpvader, 2023-08-25:
Careful here: exposing anything from a GX device, and especially things related to nodejs and thus Node-RED is a bad idea - from a security point of view.

I recommend strongly against it, even though I'm well aware that it might be quite some work to organise an alternate customisable solution to show a dashboard on the internet.

dirkjanfaber, 2024-11-08:
Closing this issue as won't solve.

closed 2024-11-08 by dirkjanfaber