victronenergy/node-red-contrib-victron #186: [Feature]Include payload rounding in 'On change' handling

pippajb, 2023-09-08:
This for me is sort of a problem, but I see it is coded this way most likely intentionally...

With the introduction of the 'On change' and 'Rounding' node configuration I was expecting to only see a new message when the rounded payload changed! This make more sense to me and would also reduce unnecessary downstream processing.

What currently happens is that any change in payload triggers an 'On change' message, regardless if after rounding the payload isn't actually seen to change.
 
Thus I would like to see the payload rounding taken into account, rather than the 'raw' data only.

dirkjanfaber, 2023-09-08:
The quick work-a-round is to put a filter node between it. But I agree that it should be handled by the node.

pippajb, 2023-09-08:
Yes I have a subflow for processing all my inputs, which discards messages with the same payload value, oh and counts msg reception too.  But as I think we agree, it would be nice for this to be handled by default.

dirkjanfaber, 2023-09-11:
Added this feature to the code and merged it into master.

closed 2023-09-11 by dirkjanfaber