victronenergy/node-red-contrib-victron #191: [BUG] Custom Control Node doesn't accept string input for string values

baroldgene, 2024-04-05:
**Describe the bug**
If attempting to update DBus settings that are string values (e.g. TimeZone) the Custom Control Node rejects text input with the error "value is not of type number".  I believe that this is DBus rejecting the attempt to save since the data sent and the type passed along with it don't match.

**To Reproduce**
Steps to reproduce the behavior:
1. Using node  'Custom Input Node'
2. Fetch `com.victron.settings` with the path "Settings/System/TimeZone"
3. Route the output directly to Custom Control Node with the same Input and Measurement.
4. Error message: "Error: Data: America/Denver was not of type number"

**Expected behavior**
I would expect to be able to properly set this setting.

**Screenshots**
[image]
[image]

**Flow**
If applicable, add a flow to help explain your problem.

**Hardware (please complete the following information):**
 - Hardware: Cerbo GX

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.5.15
 - Venus OS version: 3.30
 

**Additional context**
As best I can tell the problem lies in [this function](https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/services/dbus-listener.js#L293-L315).  I believe adding the following code after line 302 would solve the issue:
`if (typeof value == "string" && !isNaN(value)) { numType = 's' }`

baroldgene, 2024-04-05:
I was attempting to test this fix myself but was unable to due to:

1. Inability to install a custom version of the node module on the built-in NodeRed install
2. Complete unfamiliarity with how Node Red modules work

dirkjanfaber, 2024-04-06:
Thanks for reporting. I can reproduce it and it indeed needs a fix like the one you proposed. If you want to test it on your Cerbo before the fix will be part of Venus, you can do that as follows:
- first remount `/` read-write:  run `/opt/victronenergy/swupdate-scripts/remount-rw.sh`
- then edit the file `/usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/dbus-listener.js` as you already figured out and adjust it like [this](https://github.com/victronenergy/node-red-contrib-victron/commit/10169e72ff7e5392d0b1ad2b012ca1c24c4eb5f6).
- then restart node-red by `cd`-ing to `/service` and running `svc -d node-red-venus && svc -u node-red-venus` (or just rebooting the Cerbo)

Anyway, I've prepared the patch and will make a new release early next week and make it part of the next Beta Venus release.

I'll leave this issue open until that has been done.

baroldgene, 2024-04-07:
Thanks for the patch and for the help on how to test.  I was able to test this on my Cerbo GX and it worked perfectly!  Much appreciated!

dirkjanfaber, 2024-04-09:
Closing this issue; new release has been created and put on the queue for inclusion into the next Venus beta release.

closed 2024-04-09 by dirkjanfaber