victronenergy/node-red-contrib-victron #195: [ESS] Add peak shaving parameters to settings control node

inigoml, 2024-07-06:
**Is your feature request related to a problem? Please describe.**
Peak shaving options are not exposed at settings control node. There are countries like for example Spain where electric companies have different power periods, for example "high" and "low" power period. Usually low period is at night and is very cheap while high period is much more expensive. So, people select a higher power for "low period" and lower power for "high period". For example, you set a 10kW limit for "low period" in order to charge your electric car and a 5kW limit for "high period" for your regular home charges. Take into consideration that this is additional to low and high rates. It's just a power limitation.

**Describe the solution you'd like**
Exposing peak shaving settings will allow to program Victron inverter in order to enable two different power limits depending of the power period. In particular, be able to change maximum input and output power for Peak Shaving feature. 

**Additional context**
Included existing screenshot of ESS control node where no peak shaving settings are available.
Included existing screenshot of Victron Remote Control where peak shaving options can be set.
Attached simple node red flow in order to change peak shaving charge rate... when available.

[image]

[image]

[image]

dirkjanfaber, 2024-07-06:
Good suggestion. Until I get around to adding it, you can already read from and write to this using the custom input and custom output node. 

You probably want to read from and write to these values:
[image]

For the limits write `-1` to it to disable it. For changing the Peak Shaving setting either send a `1` (Always) or `0` (Above minimum SOC only).

Leaving this issue open so I don't forget adding it properly.

inigoml, 2024-07-06:
Good suggestion! Will follow your advice in the interim.
Thank you!

dirkjanfaber, 2024-09-13:
Closing as it is part of 1.5.20.

closed 2024-09-13 by dirkjanfaber