victronenergy/node-red-contrib-victron #196: VE.Smart Networking

pmknowles, 2024-07-15:
I have a Smart Shunt connected via Bluetooth to a Smart Solar MPPT 250/60 which is connected by a USB VE-Direct cable and a Raspberry Pi running Venus. There's also a RS Smart Inverter. The Smart Shunt sends the data to the charger but there doesn't seem to be a way to get it into node-red (or into the Remote Console for that matter) unless I'm missing something. 

Node-red nodes with data from the VE.Smart bluetooth network

Read the VE.Smart Networking manual

[image]

dirkjanfaber, 2024-11-08:
Did you try getting the data with https://flows.nodered.org/node/node-red-contrib-victron-vedirect-usb ?

pmknowles, 2024-11-09:
> Did you try getting the data with https://flows.nodered.org/node/node-red-contrib-victron-vedirect-usb ?

Missing the point. The data is available via the CANBus in VRM but not avaiable in the Node. I get it from the VE.Bus but not all data points are available there either.

dirkjanfaber, 2024-11-09:
Indeed getting the feeling that I am missing something here too. I am not a magician that knows the ins and outs of your system. I do know that system data in VRM is not available via the CAN bus. The only information that goes into VRM is running via the underlying dbus and posted to VRM via vrmlogger or real-time via a MQTT connection.

All of that information (on the dbus) can be fetched via the custom input node, if you know the right service and path on the dbus. Most paths are also linked using one of the more specific nodes.
Besides this, extra information on the system can also be obtained using a USB VE-Direct cable, which can be accessed using the named node-red-contrib-vedirect-usb node. Hence that suggestion.

VE.Smart networking is not available from Venus, so the information on that network can't be accessed using any node.

But that doesn't help you much, so lets try to get the data that you want to have. What data points exactly are you looking for? Please provide some screenshot clarifying it. "The data" isn't that helpful.

closed 2025-05-03 by dirkjanfaber