victronenergy/node-red-contrib-victron #197: VE.Smart Networking

pmknowles, 2024-07-15:
I have a Smart Shunt connected via Bluetooth to a Smart Solar MPPT 250/60 which is connected by a USB VE-Direct cable and a Raspberry Pi running Venus. There's also a RS Smart Inverter. The Smart Shunt sends the data to the charger but there doesn't seem to be a way to get it into node-red (or into the Remote Console for that matter) unless I'm missing something. 

Node-red nodes with data from the VE.Smart bluetooth network

Read the VE.Smart Networking manual

[image]

patrick-dmxc, 2024-10-15:
Its an Dublicate of #196

dirkjanfaber, 2024-11-08:
Closing this as a duplicate.

closed 2024-11-08 by dirkjanfaber