victronenergy/node-red-contrib-victron #198: [BUG] Victron objects are not stored homogeneously as global objects

weichenb, 2024-10-04:
I appreciate that Victron objects that are requested in a flow, get also populated by global objects/variables. I observed that majority is stored in “victronenergy” object, but a few in a “com”/“victronenergy”-object.
[image]

From my point of view it would makes sense to shift the “com”/"victronenergy-object into the “victronenergy”-object. In that case all objects would be well structured.

dirkjanfaber, 2024-10-04:
All of the nodes _should_ store it there, so there must be something wrong on [this line](https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/nodes/victron-nodes.js#L81), not converting the dbus path correctly. 

Do you happen to know which node is putting the data on the wrong place?

weichenb, 2024-10-04:
It seems that all "settings"-parameters are impacted.
Since I use only a few in my flows, I cannot provide a comprehensive statement.
[image]

dirkjanfaber, 2024-10-14:
Resolved in the latest release (which has been included in the latest beta release of Venus).

closed 2024-10-14 by dirkjanfaber

weichenb, 2024-10-18:
> Resolved in the latest release (which has been included in the latest beta release of Venus).

Thanks Dirk,
can confirm that it is solved in latest beta.