victronenergy/node-red-contrib-victron #201: [BUG] I cannot read/change the “AC import current limit (A)” of the Peak Shaving.

frbuceta, 2024-11-07:
**Describe the bug**
Cannot read or change the Peak Shaving import from the network from the NodeRed function

**To Reproduce**
Steps to reproduce the behavior:
1. Using node  'victron-{input/output}-settings'
2. I cannot select “AC import current limit (A)” which has been introduced in version 1.5.20.  e4f4fc4c9c1574c088407371b7559c16702723b6

**Expected behavior**
To be able to read and write this Peak Shaving parameter to be able to change it according to my network power which is different between hours.

**Screenshots**
[image]

**Flow**
If applicable, add a flow to help explain your problem.

**Hardware (please complete the following information):**
 - Hardware: Cerbo GX

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.5.23
 - Venus OS version: 3.51.0
 

**Additional context**
Add any other context about the problem here.

dirkjanfaber, 2024-11-08:
You are right. Strangely enough the path isn't called 'AcImportLimit', but 'AcInputLimit' instead. 
Work-around is using the custom control node instead.

dirkjanfaber, 2024-11-08:
Resolved in the upcoming version 1.6.0.

closed 2024-11-08 by dirkjanfaber