victronenergy/node-red-contrib-victron #219: [Feature] poll less frequently (or stop polling)

dirkjanfaber, 2025-03-05:
**Describe the bug**

Every 5 seconds the poller queries _all_ info from the dbus, causing an extra load of the system. This has been in the code for several years already, so apparently it hasn't been a real issue. But it would be wise to handle this in a better way. 

Relevant code is here: https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/services/dbus-listener.js#L117

Chris927, 2025-04-02:
TLDR; I did some performance profiling, and this indicates that the polling is the cause for the majority of the CPU work done by the node-red process.

Disclaimer: As is common with performance profiling, results depend on the environment and configuration of the system under test; given a different environment or another configuration, these results may not be reproducible.

The citations below are from the file produced by `node --prof-process isolate-0xnnnnnnnnnnnn-v8.log > processed.txt`, see [the NodeJS docs](https://nodejs.org/en/learn/getting-started/profiling) for more information on profiling.

The profiling run was done on a Cerbo GX, with node-red running for a few hours, without any user interaction, and one connected device (a Ruuvi temperature sensor).

## Where is time spent?

The summary shows that 85.7% are spent in shared libraries.

```
 [Summary]:
   ticks  total  nonlib   name
  194953   14.3%  100.0%  JavaScript
      0    0.0%    0.0%  C++
  294258   21.7%  150.9%  GC
  1164176   85.7%          Shared libraries
```

Under "Shared libraries", 65.4% (889,501 ticks) are spent in node. (I don't understand why node is considered a shared lib here, but that should not matter in terms of the result.)

It is worth noting that 21.7% spent on garbage collection ("GC") indicate some level of memory pressure. (This might not be a problem, yet, though.)

```
 [Shared libraries]:
   ticks  total  nonlib   name
  889501   65.4%          /data/home/root/.nvm/versions/node/v22.14.0/bin/node
  259978   19.1%          /lib/libc.so.6
   5695    0.4%          /usr/lib/libstdc++.so.6.0.32
   4584    0.3%          /lib/libgcc_s.so.1
   4155    0.3%          [vdso]
    261    0.0%          /lib/libm.so.6
      2    0.0%          /lib/ld-linux-armhf.so.3
```

Looking at the "heavy profile" next, we see where the 889501 ticks are spent. In the below excerpt, I generously removed lines that are not relevant. When analyzing the related code (mainly dbus-listener.js, related to [`_requestRoot()`](https://github.com/victronenergy/node-red-contrib-victron/blob/a9557054207102f881f1bebe5e1dd9c11a60d5d3/src/services/dbus-listener.js#L171), [`victron-client.js`](https://github.com/victronenergy/node-red-contrib-victron/blob/a9557054207102f881f1bebe5e1dd9c11a60d5d3/src/services/victron-client.js#L40) and [dbus-native-victron/lib/dbus-buffer.js](https://github.com/Chris927/dbus-native/blob/5a223d71d54b5ecf143b3f026dadd1ab898cb7c5/lib/dbus-buffer.js#L132) related to receiving messages from dbus), I can see that all of this effort is related to the polling @dirkjanfaber is concerned about.

```
 [Bottom up (heavy) profile]:
  Note: percentage shows a share of a particular caller in the total
  amount of its parent calls.
  Callers occupying less than 1.0% are not shown.

   ticks parent  name
  889501   65.4%  /data/home/root/.nvm/versions/node/v22.14.0/bin/node
  152963   17.2%    /data/home/root/.nvm/versions/node/v22.14.0/bin/node
  77083   50.4%      JS: *messageHandler /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/victron-client.js:40:28
  33642   22.0%      JS: *<anonymous> /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/dbus-listener.js:201:43
   8564    5.6%      JS: *<anonymous> /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/victron-client.js:41:24
   6648    4.3%      JS: *<anonymous> /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/dbus-listener.js:181:5
   4599    3.0%      JS: ^<anonymous> /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/dbus-listener.js:201:43
   4361    2.9%      JS: ^<anonymous> /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/dbus-listener.js:181:5
   3425    2.2%      JS: *keys /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/node_modules/lodash/lodash.js:13374:18
  127851   14.4%    JS: *messageHandler /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/victron-client.js:40:28
  81398    9.2%    JS: *slice node:buffer:636:12
  44384   54.5%      JS: *readSimpleType /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/node_modules/dbus-native-victron/lib/dbus-buffer.js:132:62
  18676    2.1%    JS: *<anonymous> /usr/lib/node_modules/@victronenergy/node-red-contrib-victron/src/services/dbus-listener.js:201:43
...
```

## What does this mean?

It means that a node-red instance doing "nothing" is spending the majority of its effort on polling. The purpose of the polling is (as I understand it) to refresh a cache. The cache is probably not used, until a user interacts with the node-red browser front end, and at the point of the user interaction, the cache may be up to 5 seconds outdated.

Further, using polling seems against the philosophy of D-Bus: D-Bus is designed to emit signals in case properties change, see section “Standard Interfaces” on emitting events in the [D-Bus specification](https://dbus.freedesktop.org/doc/dbus-specification.html). Also see the statements on signals under “org.freedesktop.DBus.Properties.PropertiesChanged”.

If we leave this as is, the work required for polling will increase with the number of devices connected (including virtual devices), and the number of settings present, etc. The risk is that this work will eventually make the system sluggish and slow to respond.

## Suggested Next Steps

[... cut ...]
This needs to be done carefully, though: Is the polling maybe used for something else, not just for the cache? Is the cache indeed used for user interactions only? If those assumptions turn out to be wrong, we have to reconsider.

Further, what if we need properties that do not support signaling? In this case, we would have to either keep a cache, or query those properties when needed. This may increase the latency perceived by the user when interacting with the front end. Maybe a cache with a different refresh strategy (not polling) is an option?

I suggest to take it step by step :)

dirkjanfaber, 2025-05-02:
This has been merged into master and release 1.6.25

closed 2025-05-02 by dirkjanfaber