victronenergy/node-red-contrib-victron #229: [BUG] Virtual Switch not visible in VRM

pontusr, 2025-04-16:
**Describe the bug**
The documentation is a bit ambiguous but it seems to indicate that the virtual switch output state should be able to be both read and set in VRM and the Remote Console. 

**To Reproduce**
Steps to reproduce the behavior:
1. Using node  'Virtual Device' set to Switch
2. Tried using different combinations of states on the SwitchableOutput. Confirmed these are correctly communicated on the Dbus.
3. No Device or Switch visible in VRM or the Remote Console
4. Virtual Device set to Tank displays immediately.

**Expected behavior**
The documentation is a bit ambiguous on this but I expect the Switch to show up under Switches in Remote Console and display the state and also allow manipulation of the state - like a relay. I also expect the state to be logged in VRM.

**Hardware (please complete the following information):**
 - Hardware: Cerbo GX

**Software (please complete the following information):**
 - node-red-contrib-victron version: 1.6.23
 - Venus OS version: v3.60~59

dirkjanfaber, 2025-04-16:
Yes, that is a known issue and we are working on that. The changes in gui-v2 haven't been merged into the Venus beta firmware yet. This is still WIP.

mtryfoss, 2025-04-30:
Do you got any estimate for when this will be available in the beta release?

mtryfoss, 2025-05-05:
Seems to be fixed now.

dirkjanfaber, 2025-05-06:
Indeed, it came out last Friday.

The feature for setting 0 switches or 0 PWM's is in https://github.com/victronenergy/node-red-contrib-victron/commit/61f3c8960c54be3aeeea9896bb571f0f95424ecb and part of 1.6.25 (which will be available in the next beta Venus).

Then the ability to alter the Custom Name. That is still under investigation. Just created [a new issue](https://github.com/victronenergy/node-red-contrib-victron/issues/232) for that.

Closing this issue now as the original problem has been resolved.

closed 2025-05-06 by dirkjanfaber