victronenergy/node-red-contrib-victron #236: [BUG]

sbender9, 2025-05-16:
**Describe the bug**
The Virtual Switch Input node sends a zero when the nodes are deployed. This can turn switches that were turned on back off.

**To Reproduce**
Steps to reproduce the behavior:
1. Using the remote console, turn a switch on.
2. In node-red, do a deployment
3. The switch gets turned off

**Expected behavior**
The node should not send any messages unless someone actually clicks a switch?

dirkjanfaber, 2025-05-16:
I am working on adding persistency to the virtual switches.

sbender9, 2025-05-16:
I think this could still be an issue even with persistency?

Say the VenusOS device is off and while it is off a switch is flipped on via some other method, then when Venus comes up it will send the last known value and turn the switch off?

dirkjanfaber, 2025-05-16:
The idea for the switch is to get 3 options for startup of the device.
- previous (from memory)
- on
- off

The slider (which is the new name for the pwm) will get:
- previous (from memory)
- off (0%)
- custom value (0-100%)

Where memory will be persistent context memory, stored to flash on interval and on re-deploy. 
I think that is about all you can expect from the virtual device. Or I am overlooking some options.

sbender9, 2025-05-16:
I would think one more option: do nothing at startup

I don't want you to send anything unless the user taps the switch on the screen. 

At startup I will send the current state so you know what it really is and the screen is correct, but when your virtual device starts up, you can't know what the actual state is. 

Just so you know what I am doing here, I have these switches actually controlling navigation lights on my boat. And there are multiple ways to control them. So if my Cerbo is down and I turn on the anchor light, I don't want the Cerbo to turn it back off when it starts up. 

Now this is very edge case and probably not what you intended this to be used for. Since I know that this can happen, I can deal with it.

dirkjanfaber, 2025-05-16:
Ok, I'll see if I can add that too. Thanks for the additional info. It is good to hear about such use cases.

closed 2025-05-20 by dirkjanfaber